ucdavis/erplab #25: Re-refrencing/Channel labeling Error

percpt, 2015-05-05:
### Reported by Justin Ruppel ###

> On May 5, 2015, at 10:07 AM, Justin Ruppel <ruppel@psych.utoronto.ca> wrote:
>
>  We recently upgraded to the most recent versions of ERPLAB(4.0.3.1) and
> EEGLAB (13_4_4b), and came across one important difference from the old
> versions that we are still trying to make sense of.
>
>  - We reference our data to channels 65 and 66 (EXG1 and EXG2)
>  - when we plotted our ERPs, we noticed that EXG1 and EXG2 were missing,
> and that we now had EXG7 and EXG8 in our plots (which we didn't have in
> previous versions)
>  - when looking at the channel list in the channel operations, it seems
> EXG1 and EXG2 were missing, and channel 65 and 66 were now labelled as
> EXG3 and EXG4.  Suggesting the entire channel mapping has been changed.
>  - we haven't changed anything in our pipeline, and have tested this with
> several different data sets.
>
>  I'm unclear whether this is a simple mislabeling, a problem with
> referencing or channel mapping, or perhaps something I'm missing in the
> processing pipeline with the new version.  Please forgive me, I know
> this is a fairly pedestrian question but I can't seem to find any
> solutions online and I'm not sure how to correct it.
>
>  Any advice or referrals to people who might be able to help would be
> greatly appreciated.
>
>
> Thanks!
>
>
> Justin

percpt, 2015-05-05:
I replied to Justing and asked for example script/data to recreate the issue.

jasonarita, 2015-05-28:
05/12/15 Justin responded

>The grad student who's been encountering the issue doesn't use a
processing script but rather uses the GUI.  Here are the steps he's been
following that lead to that point:
>
> 1) import BDF file (assign reference 65 66)
> 2) Create Event List (ERPLab > Eventlist > Create Event List > Basic)
> 3) Assign bins (these can be found in the Scene Memory folder "Binlist.txt)
> 4) Extract bin based Epochs (-200 - 1500)
> 5) ERPLab > EEG Channel Operations > This is where the error starts - EXG1
and EXG2 are not there, and 65 and 66 are now EXG3 and EXG4. Also, EXG7,
and EXG8 are listed, but these don't exist.