ucdavis/erplab #27: 2014b Plot ERPs issue

joelau1101, 2015-05-20:
Hi,

Even with the bugfix version, there seems to be problems when plotting ERPs on Matlab 2014b.
When I select ERPLAB -> Plot ERP -> Plot ERP waveforms

The following error appears:

Error using matlab.ui.Figure/get
There is no ActivePositionProperty property on the Figure class.

Error in maximize (line 119)
    oldActivePositionProperty=get(h,'ActivePositionProperty'); % an undocumented figure property

Error in ploterps (line 321)
        maximize(hbig) ;

Error in pop_ploterps (line 684)
ploterps(ERP, binArray, chanArray,  qpstyle, qMgfp, qBlc, qxscale, qyscale, qLineWidth, qisiy, qFontSizeChan,
qFontSizeLeg, qFontSizeTicks, qerrorstd,...

Error in pop_ploterps (line 301)
        [ERP, erpcom] = pop_ploterps(ERP, binArray, chanArray, 'AutoYlim', rAutoYlim, 'BinNum', rBinNum, 'Blc',
        blcorr,...
 
Error while evaluating Menu Callback

If I uncheck "Maximize Figure", it does show a graph but with only one electrode that is not labelled and there is no legend. The follow error also shows up:

No constructor 'schema.prop' with matching signature found.

Error in neozeroaxes>plistener (line 156)
        p = schema.prop(hC, 'Listeners__', 'handle vector');

Error in neozeroaxes (line 144)
plistener(axesin,'XLim',@xylim)

Error in ploterps (line 574)
                        neozeroaxes(1, fsaxtick, BCKGCOLOR)

Error in pop_ploterps (line 684)
ploterps(ERP, binArray, chanArray,  qpstyle, qMgfp, qBlc, qxscale, qyscale, qLineWidth, qisiy, qFontSizeChan,
qFontSizeLeg, qFontSizeTicks, qerrorstd,...

Error in pop_ploterps (line 301)
        [ERP, erpcom] = pop_ploterps(ERP, binArray, chanArray, 'AutoYlim', rAutoYlim, 'BinNum', rBinNum, 'Blc',
        blcorr,...
 
Error while evaluating Menu Callback

Thanks!

Joe

thimo123, 2015-06-02:
I have a similar issue with plotting ERPs in Matlab 2014b. I tried replacing "handle listener" with "addlistener" to resolve the issue as was suggested in another discussion post; however, this generated the following errors:

Error using matlab.graphics.axis.Axes/addlistener
Event 'PropertyPostSet' is not defined for class 'matlab.graphics.axis.Axes'.

Error in neozeroaxes>plistener (line 155)
hl = addlistener(hC, prp, psetact, func);

Error in neozeroaxes (line 144)
plistener(axesin,'XLim',@xylim)

Error in ploterps (line 574)
                        neozeroaxes(1, fsaxtick, BCKGCOLOR)

Error in pop_ploterps (line 684)
ploterps(ERP, binArray, chanArray,  qpstyle, qMgfp, qBlc, qxscale, qyscale,
qLineWidth, qisiy, qFontSizeChan, qFontSizeLeg, qFontSizeTicks, qerrorstd,...

Error in pop_ploterps (line 301)
        [ERP, erpcom] = pop_ploterps(ERP, binArray, chanArray, 'AutoYlim', rAutoYlim,
        'BinNum', rBinNum, 'Blc', blcorr,...
 
Error while evaluating Menu Callback

xejob, 2015-06-30:
I have the same problem running eeglab13_4_4b in Matlab 2014b on OS X Yosemite 10.10.3. When selecting Plot ERP Waveforms a plot featuring a single channel in top left hand corner with no label or legend with the following error message:

No constructor 'schema.prop' with matching signature found.

Error in neozeroaxes>plistener (line 156)
        p = schema.prop(hC, 'Listeners__', 'handle vector');

Error in neozeroaxes (line 144)
plistener(axesin,'XLim',@xylim)

Error in ploterps (line 574)
                        neozeroaxes(1, fsaxtick, BCKGCOLOR)

Error in pop_ploterps (line 684)
ploterps(ERP, binArray, chanArray,  qpstyle, qMgfp, qBlc, qxscale, qyscale, qLineWidth, qisiy,
qFontSizeChan, qFontSizeLeg, qFontSizeTicks, qerrorstd,...

Error in pop_ploterps (line 301)
        [ERP, erpcom] = pop_ploterps(ERP, binArray, chanArray, 'AutoYlim', rAutoYlim, 'BinNum', rBinNum,
        'Blc', blcorr,...
 
Error while evaluating Menu Callback

I have already made the bug fixes suggested on this link to do with matlabs new graphics system HANDLE.LISTNER:
https://github.com/lucklab/erplab/commit/92aa0d8e7897bf34f1929c13ae1a447e54132eef#diff-117ef48043c2fdde0d05f3198438c8f4

I don't know what else to do. Has there been any progress since the previous posts?

Thanks, 
Xavier

thimo123, 2015-06-30:
> 92aa0d8#diff-117ef48043c2fdde0d05f3198438c8f4 <https://github.com/lucklab/erplab/commit/92aa0d8e7897bf34f1929c13ae1a447e54132eef#diff-117ef48043c2fdde0d05f3198438c8f4>

Hi Xavier,

No, unfortunately I haven’t been able to find a solution to this yet. I have been going part way with ERPLab and then finishing with EEGLab because some of the ERPLab functions aren’t working for me. Wish I had better news for you.

Tim

diazlab, 2015-07-01:
regarding the error:

Event 'PropertyPostSet' is not defined for class 'matlab.graphics.axis.Axes'.

I was able to resolve what I believe is the same issue in an unrelated code (rotateXlabels from the file exchange), by changing all occurrences of  'PropertyPostSet' to 'PostSet'.

jasonarita, 2015-07-01:
@inheniero provided a fix for this issue in commit [b18dd09](https://github.com/lucklab/erplab/commit/b18dd090c770bdcf57c7dc3eb40841928bb5d663), as detailed in Issue #16. Try using the latest [ERPLAB 4.0.4.2 pre-release](https://github.com/lucklab/erplab/releases/tag/4.0.4.2) and see if that solves your issue.

closed 2015-08-07 by jasonarita