ucdavis/erplab #29: Measurement Tool Issue

thimo123, 2015-06-03:
When attempting to use the measurement tool to take the measure the mean amplitude between two latencies in ERPLab with Matlab 2014b, the following errors occur. Does anyone know how to resolve this please?

Mean amplitude measurement in progress...

Baseline period = 'pre' will be used for measurements

Taking measurements across ERPset #1...

Mean amplitude measurement in progress...

Output file will have extension .txt

Baseline period = 'pre' will be used for measurements

Taking measurements across ERPset #1...
Creating 1 text output file...
Error using fprintf
Invalid file identifier. Use fopen to generate a valid file identifier.

Error in exportvaluesV2 (line 156)
                fprintf(fid_values,  '%s\t', binline);

Error in pop_geterpvalues (line 747)
                        exportvaluesV2(ERP, {A}, binArray, chanArray, fname, dig,
                        k+appendfile, binlabop, foutput, mlabel, lat4mea)

Error in pop_geterpvalues (line 260)
        [ALLERP, Amp, Lat, erpcom] = pop_geterpvalues(ALLERP, latency, binArray,
        chanArray, 'Erpsets', erpset, 'Measure',...

Error in pop_geterpvalues (line 891)
                [ALLERP, Amp, Lat, erpcom] = pop_geterpvalues(ALLERP);

Error in pop_geterpvalues (line 260)
        [ALLERP, Amp, Lat, erpcom] = pop_geterpvalues(ALLERP, latency, binArray,
        chanArray, 'Erpsets', erpset, 'Measure',...

 
Error while evaluating Menu Callback

>>

jasonarita, 2015-06-03:
Try reset ERPLAB's working memory (**`ERPLAB > Settings > Reset ERPLAB's working memory`**) and see if that solves your issue.

thimo123, 2015-06-05:
Hi Jason,

Thank you very much for your reply. Unfortunately, this did not solve the issue. It seems that Matlab 2014b has changed some commands that ERBLab was using, creating some bugs. The rest of ERP Lab is working great for me. I am still trying to debug the “ERP Measurement Tool” and the “Plot Scalp ERPs”. I am still a Matlab beginner though so this may be beyond me.

Best,

Timothy

jasonarita, 2015-07-01:
@thimo123

Could you provide some more information on how you arrived at this error. What input values did you set in the `ERP Measurement Tool` when running it? That'll help me recreate your error so I can look into it. 

For example, what were the values for the following paramenters:
- Type
- Interpolation Factor
- Measurement Window
- Baseline period
- Precision
- etc.

You could even just drag and drop a screen shot of the tool like the one below:
[image]

----
Just from the error message you provided alone, it seems the cause is coming from writing to the output text-file in the `exportvaluesV2` function.

```
Creating 1 text output file...
Error using fprintf
Invalid file identifier. Use fopen to generate a valid file identifier.

Error in exportvaluesV2 (line 156)
fprintf(fid_values, '%s\t', binline);
```

So, somehow the file identifier, `fid_values`, variable got corrupted or maybe was never correctly generated in the first place. One possibility, is that you don't have write permissions the the save directory. [Here's a discussion of invalid file identifier error on Stack overflow.](http://stackoverflow.com/questions/10606373/what-causes-an-invalid-file-identifier-in-matlab)

What would really help, would be to see the value of the `fid_values` variable. If you're familiar with Matlab's debugger you can set a breakpoint at line 156, run the tool, and then access the `fid_values` variable and tell me what it is set to at the time of the error. 

Otherwise, If you add the following code in `exportvaluesV2.m` on line 155 (right before the `fprintf(fid_values,  '%s\t', ...` call)

```
fprintf('File identifier: %d \n', fid_values)
```

This should print out the the value to your command window, and from there you can tell me what the file identifier number is.