ucdavis/erplab #31: Bug with pop_syncroartifacts?

sgbalogh, 2015-08-19:
Hi,

I keep running into a problem with pop_syncroartifacts. I am trying to use this function in a script that processes datasets, but whenever I add a line, such as:

`EEG = pop_syncroartifacts(EEG, 2);`

... I get these errors:

`Error using pop_syncroartifacts`
`Expected a string for the parameter name, instead the input type was
'double'.`

`Error in pop_syncroartifacts (line 111)
p.parse(EEG, varargin{:});`

`Error in parse_apart_bins (line 14)
EEG = pop_syncroartifacts(EEG, 2);`

Strangely enough, whenever I perform the same action via the GUI (transfer EEG.reject marks to EVENTLIST), it works perfectly –– and I always see in the console that the equivalent command was the exact one that I try in my script. Any suggestions? Thanks! -s

CEMiller, 2015-09-14:
I'm having the exact same issue. It would be great to know if anyone found a solution for this!

H-Kadel, 2016-03-29:
Hi there,

I just encountered the same problem. I can't figure out the exact reason from the code, but I found a workaround that solved it for me: Simply use

`EEG = pop_syncroartifacts(EEG,'Direction', 'erplab2eeglab')`

instead of the other code. At the place of 'erplab2eeglab', you can use the other keywords, as I figured from pop_syncroartifacts.m:

`if direction==1      % erplab to eeglab synchro`      
`dircom = 'erplab2eeglab';`                
`elseif direction==2  %eeglab to erplab synchro`   
`dircom = 'eeglab2erplab';`   
 `       elseif direction==3 % both`   
`                dircom = 'bidirectional';`   
`        else`   
`                dircom = 'none';`   

Maybe that helps?

sgbalogh, 2016-03-29:
@H-Kadel thanks for sharing this workaround!