ucdavis/erplab #57: GUI problem when creating advanced eventlists

MegaCurry, 2017-01-17:
#GUI problem when creating advanced eventlists

The GUI cannot show all the options(shown below). But other GUIs work well. After I have reinstalled my MATLAB, it's still not working.
[image]

Thank you for your help.

|   |   |
|----| ---- | 
| OS version     | [*Win10(64bit)*] |
| Matlab version | [*2015b*] |
| EEGLAB version | [*13*] |
| ERPLAB version | [*fill in*] |
GUI problem when creating advanced eventlis

andrewxstewart, 2017-01-17:
Hi MegaCurry - thanks for letting us know.

Weird - it looks like the top of that GUI is getting clipped off. I'll try to replicate this on a Windows box.

MegaCurry, 2017-01-21:
I found that GUI problem of ‘creating advanced eventlist’ still exists in ERPLAB6.1 on EEGLAB13.6.5. After I deleted my EEGLAB13.6.5 and installed EEGLAB13.5.4. It worked well. 
Thank you so much.

andrewxstewart, 2017-02-03:
Thanks for your info. I'm glad you got it sorted!

I've tried Windows 10, Matlab 2015a, EEGLAB 13.6.5 and ERPLAB 6.1.1, and the Advanced Eventlist GUI seemed to render correctly.

Since the new EEGLAB folder did fix this for you, I suspect that some old text size options might have caused this. I've made new versions a little more robust against this.

If anyone else has this issue, try:
- resetting the ERPLAB memory, or
- reinstalling the most recent ERPLAB in a fresh folder path

[image]

closed 2017-02-03 by andrewxstewart