ucdavis/erplab #62: Documentation: API Reference and PDF

cogmind, 2017-04-03:
### Description

There is no PDF of the wiki documentation and there is no API Reference Manual in the wiki documentation. Docstrings appear to exist for all functions, but have not been "published".

jasonarita, 2017-04-06:
Thanks, we're looking all our options for publishing them.

andrewxstewart, 2017-04-07:
Hi Daniel,

I'd like to have a single PDF of the wiki material too. I'm looking at making that now.

For the docstrings, the function headers have been written to be easily readable from within Matlab. For instance, you can type 'help pop_erlabShiftEventCodes' or 'help functionName' at the Matlab command window and the functions header help will be produced at the Matlab command window.

Additionally, I'd also like to have a function reference list outside of this, perhaps as a table in the wiki, with a list of all functions and the corresponding help info.

Is this what you are looking for?

cogmind, 2018-02-16:
Have you considered using a document generator?

[Doxygen](http://de.mathworks.com/matlabcentral/fileexchange/25925-using-doxygen-with-matlab) (PDF Latex option)
[sphinx](https://pypi.python.org/pypi/sphinxcontrib-matlabdomain)
[m2html](https://www.artefact.tk/software/matlab/m2html/)