ucdavis/erplab #63: ERPlab Measurement Tool GUI problem

anicampose, 2017-04-05:
### I´m trying to run a "Mean Amplitude between fixed latencies" measure in the erplab´s (v 6.1.3) Measurement Tool GUI but it doesn´t let me access to the upper area of the pop-up window. I can load the erp sets but I can´t see, nor click the analysis menu that it is under the upper banner. I´ve restarted eeglab (v 14) and MAtlab (2016b) and seth the path again for eeglab, erplab, Biosig and Mass Univariate toolbox but it keeps on with this error. I´ve also reset erplab working memory and changed the screen settings but nothing seems to work.

[*Description of the bug or feature*]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*2016b*] |
| EEGLAB version | [*v14*] |
| ERPLAB version | [*6.1.3*] |

andrewxstewart, 2017-05-03:
Hi anicampose,

Which menus do you need to set? What exact options are you not able to access?

For 'Mean amplitude between two fixed latencies', the interface looks like this for me, and I think that is as intended:

[image]

Aymericccc, 2017-05-17:
Hello, 

I think I have the same problem, which is the window can't be seen in its totality.
Practically speaking, it's impossible to see the measurement type menu, therefore impossible to change the way I want to see the amplitude...

Is there any mean to solve this problem?

anicampose, 2017-05-17:
I solved it by installing an older version of ERPlab and it worked perfectly!

Aymericccc, 2017-05-17:
I just tried with the version 6.1.2 but the same problem was here, then the version 5 and it worked! Thanks for sharing!

Just by curiosity, which older version did you install?

anicampose, 2017-05-17:
V.5 as well!
Best,

Ana

andrewxstewart, 2017-05-23:
Thanks for reporting a work-around for this!

I'd still like to get this fixed in the new versions, if possible.

Do either of you have a screenshot or photo of the problem? I'm not sure
what difference you are getting in v5 or v6.1.3

Thanks

anicampose, 2017-05-23:
Hi Andrew,
I got this, and a friend got it as well, with the V6.1.3, which suggests that was not my computer or some setting that I defined, but something with the erplab version.
Best,

Ana

Envied desde mi iPhone

andrewxstewart, 2017-05-23:
Strange - I will have another look.

Can you take a screenshot or photo of the problem? And say what you wanted
to do, and what wasn't working?

Aymericccc, 2017-05-23:
Hi Andrew, 

I think we are talking about the same problem with the measurement tool. I added the screenshot so that it'll be more clear.
Once I open this tool with the V6.1.3 of erplab (and not with V5), I got this, with no mean to scroll up the window. But the I can click on the options on the screen, so basically the tool works, only the scrolling seems to have trouble.
I also tried to change the resolution of my screen, but it didn't change anything.

I hope this will help!

[image]

andrewxstewart, 2017-05-27:
Thanks again for reporting this.

I was able to recreate this problem on Windows 7 using a lower screen resolution.

I think when the extra 'Fractional Amplitude' tool were added to this window, it got a little taller. This made the window over-crowded on some screens.

I've tidied this up and fixed this now. 

Try this:
https://github.com/lucklab/erplab/archive/master.zip

Let me know if this works for you or not.

andrewxstewart, 2017-06-05:
With the latest commit, the Measurement Window is confirmed working on lower resolution screens.

This fix will be included in the next release.

[image]

closed 2017-06-05 by andrewxstewart

anicampose, 2017-06-05:
Cool! Thanks!!
Best,

Ana