ucdavis/erplab #68: ERP Measurement tool GUI cutting off

mcweenysean, 2017-07-13:
Hi,

I have run into a problem in which the ERP Measurement Tool GUI cuts off the top part of my GUI so that I am unable to select mean amplitude between two frequencies. This is one of two GUIs that this happens to with the other being the epoch4avgGUI under the weighted averager GUI. I tried changing my screen resolution from 1280x800 (best) to no avail. Are there any solutions? Attached is what my GUI looks like
[image]

|----| ---- | 
| OS version     | [*OSX 10.10.5*] |
| Matlab version | [*2016b*] |
| EEGLAB version | [*14.0.0*] |
| ERPLAB version | [*6.1.3*] |

andrewxstewart, 2017-07-13:
Hi there,

This has been recently fixed, but the fix is not yet in a full release version.

Get the latest commit, which contains the ERP Measurement Tool GUI fix, here:
https://github.com/lucklab/erplab/archive/master.zip

There are some more details here:
https://github.com/lucklab/erplab/issues/63

mcweenysean, 2017-07-13:
That worked - thanks so much!

closed 2017-07-13 by mcweenysean