ucdavis/erplab #72: issue with pop_editeventlist.m

aschetti, 2017-09-11:
### Description
Hi,
I am having troubles with the function pop_editeventlist.m (either when called via pop_creabasiceventlist.m or used directly). The error message reads:

Error in pop_editeventlist (line 190)
p.addParameter('List'                , ''        , @ischar);
No appropriate method, property, or field addParameter for class inputParser.

<br>
#### Steps to Reproduce

    % import .bdf file. Imported channels: 1:70 (64 EEG + 4 EOG + 2 mastoids)
    EEG=pop_biosig('D:\OneDrive - UGent\FSAReward\data\EEG\Exp1\VP01.bdf'],'channels',[1:70],'ref',[69 70],'refoptions',{'keepref' 'on'}); % referenced to average mastoids

    % create event list
    EEG=pop_creabasiceventlist(EEG,...
        'Eventlist','D:\OneDrive - UGent\FSAReward\scripts\EEG\Eventmapping.txt',...
        'BoundaryNumeric',{-99},...
        'BoundaryString',{'boundary'},'Warning','off');

Here is where I get the error message.
Note that:
* changing directories (e.g., by eliminating the spaces in the path) did not solve the issue;
* using the GUI did not solve the issue;
* this script used to work with the same dataset when using older versions of EEGLAB (13.6.5) and ERPLAB (sorry, can't retrieve version info... perhaps 4.0).

<br>
#### Expected behavior:
I expect the function to create an event list that I can later use to assign bins...

<br>
#### Actual behavior:
... but it crashes:

Error in pop_editeventlist (line 190)
p.addParameter('List'                , ''        , @ischar);
No appropriate method, property, or field addParameter for class inputParser.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Windows 7 SP1] |
| Matlab version | [8.0.0.783] |
| EEGLAB version | [14.1.1b] |
| ERPLAB version | [6.1.4] |

By the way, thanks for the great work... it's immensely helpful.

Greets,
Antonio

andrewxstewart, 2017-09-11:
Thanks for reporting this.

Is that Matlab version R2012b? Do you have an easy way of updating to a more recent Matlab? Places with network licenses often have easy ways of updating.

aschetti, 2017-09-11:
Unfortunately I can't update... do you already know what the cause might be?

andrewxstewart, 2017-09-11:
I think so, yes.

We do make use of both inputParser and it's sub-class, addParameter. I think addParameter was added to Matlab in R2013b.

https://www.mathworks.com/help/matlab/ref/inputparser.addparameter.html

Previously, old versions of ERPLAB used addparamvalue, but Matlab advises no longer using this.

You could try getting a newer Matlab, or using an old version of ERPLAB. Or, if you are feeling adventurous, you could try running 'edit pop_editeventlist.m', and then reverting usage of the addparameter to the old addparamvalue.

https://www.mathworks.com/help/matlab/ref/inputparser.addparamvalue.html

aschetti, 2017-09-12:
Damn. Ok, thanks for the input.
Reverting back to *addparamvalue* would be time-consuming for me at the moment. For now, I would rather go back to an older ERPLAB version... could you please give me a direct link to a version that you know does not have such a problem? I guess it's not on on GitHub... I can only see version 6 and above.

andrewxstewart, 2017-09-12:
ERPLAB 6.1.3 should work for you.

We have old releases still up on GitHub at:
https://github.com/lucklab/erplab/releases

aschetti, 2017-09-13:
Hi Andrew,
reverting to ERPLAB 6.1.3 solved the problem. Thanks so much for your help!

Greets,
Antonio

closed 2017-09-13 by aschetti