ucdavis/erplab #78: Append ERPsets failed

carolchuchu, 2018-01-17:
try to append ERPsets but the button failed like this
[image]

has anyone got the same problem? or is there any better ways to subtract ERPsets?
 
| OS version     | [macOS 10.13.1] |
| Matlab version | [R2015a] |
| EEGLAB version | [13.5.4b] |
| ERPLAB version | [v5.1.1.0] |

carolchuchu, 2018-01-17:
[image]

64Zone, 2019-07-22:
Hi!

I run into the same issue and wonder if you have solved it? Please let me know!

| OS version | [macOS 10.14.5] |
| Matlab version | [R2017a] |
| EEGLAB version | [14.1.2b] |
| ERPLAB version | [v7.0.0] |

carolchuchu, 2019-07-24:
Hi 64Zone!
It has been a long time since I run into this problem.I guess I have solved it or skipped it long ago while I can not remember. But for now, as I reread this problem, I would suggest you to check bins, channels,  sampling rate or any characters like these for each ERPset, as mismatch in format related issues are usually reasons for failures.

64Zone, 2019-07-24:
Thanks for your fast response carolchuchu!

I actually can have the GUI pop up properly in Windows but not in Mac. Does this ring a bell? Anyway, I can find a way to make it work. It just bugs me not knowing why the GUI would not run properly in Mac.

andrewxstewart, 2019-07-25:
I think this error might be from trying to print a list of ERPset files to the Append GUI, but hitting a bug if the paths are no longer valid.

As a test and possible workaround, you could try loading the desired ERPsets before running that GUI.

64Zone, 2019-07-25:
Thanks for the suggestion andrewxstewart. I tried your suggestion but the problem persisted. I feel the Mac version is really not as reliable as Windows/Linux.

klgerwin, 2019-08-29:
I am having this same issue on Windows. Any updates?

SamRTliu, 2020-03-17:
> I am having this same issue on Windows. Any updates?

Hi klgerwin,

you can try to reset the ERPLAB`s working memory from the ERPLAB menu [**settings --> ERPLAB memory setting-->ERPLAB`s working memory**]. 

Ray.

DouweHorsthuis, 2021-06-16:
I encountered the same problem and @SamRTliu 's solution solved it for me.