ucdavis/erplab #87: erpsets and plot erp  function show up

sriazbasha, 2018-08-14:
### Description

[*Description of the bug or feature*]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |
| ERPLAB version | [*fill in*] |

andrewxstewart, 2018-08-16:
Do you mean the ERPLAB menu items are blanked out in EEGLAB? They should become selectable once you have a ERPset loaded.

Do you have a loaded ERPset? Or a EEGset loaded up in EEGLAB that you can average?

How far can you get in the the tutorial before you hit a problem?
https://github.com/lucklab/erplab/wiki/Tutorial

sriazbasha, 2018-08-16:
Thanks for your response . I have done till compute avg ERPs and did try to
plot ERP which does not get selectable to plot . Even tried to load a
erpset  directly and see if it works that way. Both ways both erpset menu
item and plot erp are inactive and donot go in  selctive mode. Appreciate
your help.Thanks

sriazbasha, 2018-08-16:
> Ok cool.
>
> What happens after you run 'Compute Averaged ERPs'?

I am able to generate erp file  after compute avg erp and save it and after
that the erpset menu item or plot erp are still not available and donot
work

andrewxstewart, 2018-08-16:
Can you email me a screenshot of the problem, and the ERP file you have saved and loaded? axstewart 
 ucdavis edu

One thing to try might be to clear your Matlab path of EEGLAB, and then add a fresh install of EEGLAB, with a fresh install of ERPLAB in the plugins folder. Also, check that you install folder is one your Matlab definitely has write access to, like in you Documents folder.

sriazbasha, 2018-08-16:
Will do so. Hope will try to do it from the scratch

andrewxstewart, 2018-09-04:
Did this resolve the issue, Riz?