ucdavis/erplab #91: Major Bug(s) in Epoch Subset Assistant

mcweenysean, 2018-08-31:
Hi all,

I've been using the epoch subset GUI (in the averager GUI) for several months and I think I found a pretty serious issue with how it subsets even and odd epochs as well as proportions using relative data. I looked at ERP.ntrials.accepted for each of my splits (how I created the splits is in the screenshots [image]
[image]
[image]
), and found some inconsistencies that would suggest that the splits are not working correctly. For example, in even_odd_forerplab
[even_odd_forerplab.xlsx](https://github.com/lucklab/erplab/files/2340518/even_odd_forerplab.xlsx)
, in the even split, subject 1203 has 43 epochs in bin 2 and 40 in bin 5. In the odd split, they have 44 in bin 2 and 40 in bin 5. While this seems great, there are only 40 TOTAL artifact-free bins 5s. This is evidenced not only by our “Summarize Artifact Detection Marks in a Table,” but also when looking at split halves by time (next paragraph). This suggests to me that the bin 5 was never split, and when I correlated even and odd splits, they were much higher than my other splits. 

The other concerning issue is how the epoch subset function deals with proportions of data with relative bins. In the attached spreadsheet (qs_forerplab: [qs_forerplab.xlsx](https://github.com/lucklab/erplab/files/2340519/qs_forerplab.xlsx)) for subject 1203 has 54 epochs in bin 2 and 12 in bin 5 for q1 (screenshot for how that was generated is attached) while they have 33 epochs in bin 2 and 28 in bin 5 for q2 (screenshot also attached). Despite me using the “relative (referred to each bin’s epochs)” option, it seems that it created an absolute split. 

While the latter bug (seemingly using absolute instead of relative) is consistent, the even-odd splitting bug for bin 5 is inconsistent (approx 50% of the even-odd split halves have the correct number of trials in bin 5). Even worse, it doesn’t seem to be bin specific, as bin 2 has one subject for whom the data was not properly split (4828). I have gone through the GUI again for several subjects to confirm that the GUI is still generating the same numbers for each participant and it seems to be doing so. These trial accepted numbers seem to be accurate, as when I include only the individuals whose data was split correctly, my correlation falls considerably (~.4). Can these bugs be fixed? I'm happy to send .set and .erp files as well.

|   |   |
|----| ---- | 
| OS version     | [High Sierra] |
| Matlab version | [2018a] |
| EEGLAB version | [14.1.2] |
| ERPLAB version | [7.0.0] |

andrewxstewart, 2018-09-04:
Hi Sean,

Thanks for reporting this, and documenting the issue. Hmm - this GUI might need overhauled.

Do you see any inconsistencies in sets with:

1. no artifacts?
2. no time range or proportion specified?

Perhaps there are weird interactions with the time range, proportion, indexing method, and artifact status. This will need to be tested.

Your .set and .erp files would be welcome.

mcweenysean, 2018-09-05:
1. The inconsistencies seem to disappear when I run the GUI with no artifact marks set - of my 120 items in bin 5, each split had 60, so that looks promising. It must be interacting strangely with the artifact marks.

2. The even-odd errors were with no range specified (i.e. the whole .set), so yes, to answer question 2.

Kaip1216, 2019-02-07:
Anyone knows about the status of the odd-even epochs split bug? Has it been fixed?