ucdavis/erplab #97: Where can I download the ERPLAB tutorial?

xp752926566, 2018-11-22:
### Description

[*Description of the bug or feature*]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |
| ERPLAB version | [*fill in*] |

andrewxstewart, 2018-11-29:
Hi there,

You can see the ERPLAB Tutorial here:
https://github.com/lucklab/erplab/wiki/Tutorial

I recommend reading it there.

If you really need to download it (rather than reading it there), perhaps to read it offline, you might be able to do so by getting the Git Wiki from:
https://github.com/lucklab/erplab.wiki.git

I think a text reader like Atom or VS Code might be able to parse the markdown in the *.md files in there. But reading it on that above website is probably better.

Does that help?

xp752926566, 2018-12-01:
oh，thanks very much .

closed 2020-04-28 by andrewxstewart