ucdavis/erplab #98: the eeg.event.field contain subfield name reserved for erplab

Xinzhu-Fang, 2018-11-28:
### Description

[*I am getting this warning with two choices given: continue as it is or overwrites.  What does this mean? What I am supposed to do*]

<br>
#### Steps to Reproduce

1. Poped up when trying to create an eventlist

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Win10*] |
| Matlab version | [*2018a*] |
| EEGLAB version | [*14_1_1*] |
| ERPLAB version | [*0_7_0*] |

Xinzhu-Fang, 2018-11-28:
Also, the eventlist created is messy. I tried both "overwrite" and "continue as it is".
[image]
This is what I expect
[image]

andrewxstewart, 2018-11-29:
Hi Xinzhu-Fang,

Typically, EEGLAB has EEG-set datasets with some event info in EEG.event.*
If this is a new EEG-set, it typically won't have fields in there that collide with ERPLAB's event naming convention.

If you are running something like generating new EVENTLISTs, it can be that ERPLAB writes data to EEG.event.

If your aim is to analyze data in ERPLAB, I think you are okay to overwrite EEG.event. But be aware that this will change the EEG set in memory. As always, save important data before potential changes.

closed 2018-11-30 by Xinzhu-Fang