ucdavis/erplab #102: Incorrect event durations in EEG.epoch after applying both pop_creabasiceventlist and pop_epochbin

chsquare, 2019-02-11:
### Description

Applying 'pop_creabasiceventlist' and then 'pop_epochbin/pop_epoch' leads to wrong event durations in 'EEG.epoch' if the dataset sampling rate is different from 1000 Hz and the event duration values > 0. The reason for this behaviour is probably the following: EEGLAB stores event durations in samples/points and pop_epoch (called also from within pop_epochbin) converts the values in the duration field from samples to milliseconds. pop_creabasiceventlist does the same. Thus, applying both leads to incorrect values. Note that applying both functions, first pop_creabasiceventlist and then pop_epochbin/pop_epoch, is the default procedure when working with ERPLAB.

<br>
#### Steps to Reproduce

1. Load some non-epoched EEGLAB dataset (a "set" file) with events of duration > 0 and a sampling rate ~= 1000 Hz.
2. Apply pop_creabasiceventlist()
>> EEG = pop_creabasiceventlist( EEG );
Say event 1 has non-zero duration, check this value
>> EEG.event(1).duration
3. Run the 'binlister' in a way that event 1 ends up in a specific epoch, say the first one
4. Epoch the dataset, say from -200 to 800 ms
>> EEG = pop_epochbin( EEG, [-200, 800]);
5. Check the same event's duration in the EEG.epoch structure
>> EEG.epoch(1).eventduration

<br>
#### Expected behavior:
The absolute duration of an event in the EEG.epoch structure (given in milliseconds) should be the same as the duration of an event in the EEG.epoch structure (given in samples/points).

<br>
#### Actual behavior:
If 'pop_creabasiceventlist' was applied to an EEG dataset with sampling rate ~= 1000 and events with duration > 0, 'pop_epochbin/pop_epoch' makes the duration of the events in the EEG.epoch structure different from the duration of the events in the EEG.event structure (considering that both structures store values in different units samples/pointes versus milliseconds).

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Mac OS 10.12.6 |
| Matlab version | 9.1.0.441655 (R2016b) |
| EEGLAB version | 14.1.1b |
| ERPLAB version | 7.0.0 |

bradfde, 2021-03-16:
Did you ever get any word on this?

chsquare, 2021-03-16:
No. I'm aware of the issue, so it was fine for me. But I guess it would still be a good idea to fix that.