ucdavis/erplab #104: Pop-ups in loops

Allexxann, 2019-03-04:
### pop_epoch2continuous has insuppressable pop-up, which breaks all attempts to loop ше in scripts.

Function pop_epoch2continuous has no keys to suppress question box, you have to comment corresponding lines.

<br>
#### Steps to Reproduce

1. create for-loop with pop_epoch2continuous
2. get pop-up at the first file to process

#### Expected behavior: Expected some obvious way of suppressing pop-up, like 'warning' key.
#### Actual behavior: No such thing.
<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Win10*] |
| Matlab version | [*2017a*] |
| EEGLAB version | [*14.1.2*] |
| ERPLAB version | [*7.0.0*] |

RenzoLanfranco, 2020-04-24:
Hi, I was looking for the same thing. 

pop_epoch2continuous.m needs a parameter that allows suppressing the question box. So far, the only solution I've found is changing the erplab M file, which isn't a good one since it prevents from sharing pipelines/scripts.

andrewxstewart, 2020-04-27:
Hi there,

Thanks for the report.

I think a call like this should work with no popup and no warning:

`EEG2 = pop_epoch2continuous(EEG,'Warning','off');`

Does that work for you?

Allexxann, 2020-04-28:
Shouldn't that be in the help part of the function? Because it seems like noone here had any idea that's an option at all.

RenzoLanfranco, 2020-04-28:
Thanks Stuart. Yes, that solves the issue. -- From: Andrew X StewartSent: 28 April 2020 00:48To: lucklab/erplabCc: Renzo Lanfranco; CommentSubject: Re: [lucklab/erplab] Pop-ups in loops (#104) Hi there,Thanks for the report.I think a call like this should work with no popup and no warning:EEG2 = pop_epoch2continuous(EEG,'Warning','off');Does that work for you?

Allexxann, 2020-04-28:
Okay, this is solved, the magical parameter is here:
pop_epoch2continuous(EEG,'Warning','off');

Shame it's not mentioned in the help part of the function, but whatever.

closed 2020-04-28 by Allexxann

andrewxstewart, 2020-04-28:
I agree that this parameter should already be described in the help for this function.

I have added this now, in the latest commit. This will be included in the next ERPLAB release.

See the changes here:
https://github.com/lucklab/erplab/commit/440b0e04e169cb17900c8fe1eb959a1759d194b3