ucdavis/erplab #105: Discrepancy between ERP plots in ERPLAB and WinEEG

dkleeva, 2019-03-06:
**Description**

Hello! 

I process EEG data in ERPLAB. This data was recorded via WinEEG, where I can also perform some processing and plot ERP waveforms. The problem is that when I do the same procedure in ERPLAB I get completely different plots. It seems that there is a ~1000 ms delay and the plots in ERPLAB are shifted. The events themselves and their latencies are imported correctly, the bin-based epochs also don't contain any kind of delays.  

Here you can see the plots from ERPLAB:
[image]

And these are the plots from the original app (the dotted line is the stimulus onset):
[image]

Is there any way to solve this problem without the manual shifting of ERPs afterwards?

Thank you,

Daria.

| Matlab version | [R2018b] |
| EEGLAB version | [v14.1.2] |
| ERPLAB version | [7.0.0] |

andrewxstewart, 2019-03-06:
Hi Daria,

Thanks for the report.

I'm not familiar with WinEEG. What is the time period in the x-ticks in those WinEEG plots?

That seems very strange that the bin-based epochs and the events have okay timing, but the plotted ERPs then do have timing issues.

Let's get that sorted out.

Can you send me the EEGsets, ERPsets, and bin-descriptor that you use? You can email erplabtoolbox@gmail.com

We do have some ERPLAB reproducibility error-checks, where the timing in ERPLAB and other software broadly appear to agree. But not all cases are covered, and if there are issues with some experiments, or with data from other software, I want them addressed.

Best,
Andrew

andrewxstewart, 2019-03-28:
Hi Daria,

Thanks for sending me your data.

Looking at that EEG set, I can see you have many stimulus event codes for each trial. That’s good, but also means we need to lock to the right one for timing.

In ERPLAB, the bin-epochs are extracted (and averaged together) in temporal reference to a timelocked ‘anchor’ event code.

With the Bin-Descriptor File (BDF) you supplied, I can see you chose to ‘lock’ to event code ‘1’, with:

bin 1
Test
.{1}

The event codes immediately after the dot will become the zero-time, 0 ms, for that bin. By convention, ERP researchers typically take this zero-time to be the time of first stimulus appearing on the screen.

It could be that, for your data set, you want to ‘lock’ to another stimulus event code, in order to have the timecourse match.

Which event code is it that indicated first stimulus appearing on screen in your experiment?

With the big transient around +150ms in your WinEEG data, I suspect the stimulus time is probably correct in there. You will need to make sure that you specify this as the anchor event in your BDF text file, so ERPLAB ‘knows’ what the corresponding 0 ms event is desired for that bin.

For BDF spec tips, see:
https://github.com/lucklab/erplab/wiki/Assigning-Events-to-Bins-with-BINLISTER
https://github.com/lucklab/erplab/wiki/BDF-Library

Best,
Andrew Stewart
ERPLAB Team

andrewxstewart, 2019-03-28:
Here is an example, locked to event 100, 101, and 102:

[example_bdf.txt](https://github.com/lucklab/erplab/files/3019902/example_bdf.txt)

closed 2020-02-17 by dkleeva