ucdavis/erplab #106: selecting eventlist

ra75763, 2019-04-19:
Hi,
Can some one help me ? From the attached figure, how can I select my required event ? I want to select two signal frequent and rare. But I am confused how to do . Please help.
[upload.docx](https://github.com/lucklab/erplab/files/3096950/upload.docx)

andrewxstewart, 2019-04-19:
Hmm - those event codes look weird.

Here is an example of continuous EEG data with the frequent stimuli presentation indicated with a '22' event code, and rare stimuli indicated with a '12' event code, and subject responses coded with a '9'.

[image]

What is in your EEG.events structure? How were you registering your events? What recording amplifier are you using?

ra75763, 2019-04-19:
I am using raw data. Didn’t modify anything.  Please see my previous
email.

ra75763, 2019-04-19:
Hi , thanks for reply. I am very new user. I may be wrong. Please advice .
May be my way are wrong.

1. I am using headband and OpenBCI for getting signal. Then by openVibe
designer, I am adding a tone signal (in.wav format ). Then tries to hear
the signal and record my eeg signail with openvibe designer in CSV format.
Then trying eeglab. Please see the figures:
This is openvibe recording : I select sound by sound player. (Attached in
Doc file)

2. Then by keyboard stimulator,  I press 'a' and sound starts .Pressing
'z', sound stops.
3. Attached csv file is my recording eeg.
4. By eeglab, added channel and events.

Then just started "erplab". I am following Steve Luck video tutorial.
https://www.youtube.com/watch?v=cy7a4B0zOSU

I may be wrong. My recorded signal has two frequency Beeps. Beep1 and
Beep2. One is regular and one is rare. I have attached the tone files. "
t1.wav "
I want to do a simple oddball paradigm test. Want to plot P300.

Thanks. Please see the zip files.

Rafia

ra75763, 2019-04-20:
[New folder (3).zip](https://github.com/lucklab/erplab/files/3100138/New.folder.3.zip)
Please see the attached files.