ucdavis/erplab #107: Warning message when starting EEGLAB

arnodelorme, 2019-05-02:
### Description

I get this warning message when starting EEGLAB
"Warning: The value of local variables may have been changed to match the globals.  Future versions of MATLAB will require that you declare a variable to be global before you use that variable."

This is due to ERPLAB code in the eeg_global function of EEGLAB. How can we remove the warning message (I am maintaining EEGLAB). 

    if exist('eegplugin_erplab.m')
        global ALLERP; % Javier Lopez-Calderon for ERPLAB
    end

<br>
#### Steps to Reproduce

1. Install EEGLAB and ERPLAB
2. Start EEGLAB

<br>
#### Expected behavior:
No Warning

<br>
#### Actual behavior:
Warning is displayed

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | OSX |
| Matlab version | 2018a |
| EEGLAB version | 2019_0 |
| ERPLAB version | v7.0.0 |

andrewxstewart, 2019-05-07:
Hi Arno,

Thanks for reporting this.

Is that snippet from eeglab.m?

Strangely, I'm on (the latest?) EEGLAB v14.1.2 and ERPLAB v7 and I don't get that error.
Also Matlab R2018a.
[image]

andrewxstewart, 2019-05-09:
Ah - I see. 2019.0 is the latest development version of EEGLAB you are working on.

I do now see this error. I think it only shows once, and then Matlab needs to be restarted to see it again.

This error is produced when eeglab.m calls eeg_global.m.

Error only manifests:
- on first run of eeglab after restarting Matlab
- at line 1100 of eeglab, on the eeglab('redraw') call

I think this is the first time that eeglab is called again after eegplugin_erplab.m is run.

So I think I can fix this by ensuring that the first use of ALLERP in eegplugin_erplab.m is of the global ALLERP.

Testing this looks like it stops that Matlab warning message!

Could you check that the ERPLAB in this commit fixes this for you?
https://github.com/lucklab/erplab/archive/db4a9ca6fd74e070d1f97c5e286e5bb2b970c94b.zip

Cheers,
Andrew

arnodelorme, 2019-05-13:
Yes no more warning.

closed 2019-05-13 by arnodelorme