ucdavis/erplab #111: eegchanoperGUI cannot create a new dataset without "nchan" syntax.

elenageangu, 2019-07-08:
### Description

I am trying to interpolate bad channels with the surrounding channels using 'channel operations'. I typed in the formula following the recommendation in the manual (e.g., ch39 = (ch29 + ch38 + ch48 + ch40) / 4 ). When I click 'Run' i get the error message 

"Error: eegchanoperGUI cannot create a new dataset without "nchan" syntax."

Any help in solving this is much appreciated.
Elena

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |
| ERPLAB version | [*fill in*] |

andrewxstewart, 2019-07-25:
The syntax can be weird for 'channel operations'.

If you want to create a new, novel, dataset, have the 'Mode' set to 'Create new dataset', and then have each 'new' channel specified like:
nch1 = (ch29 + ch38 + ch48 + ch40) / 4 )

Alternatively, here you probably want the 'Mode' set to 'Modify existing dataset'. You are only changing one channel.
ch39 = (ch29 + ch38 + ch48 + ch40) / 4 )

[image]

elenageangu, 2019-07-25:
Dear Andrew,
many thanks for help with this.
Best wishes
elena

closed 2020-04-28 by andrewxstewart