ucdavis/erplab #113: Error - Moving window peak-to-peak threshold

psyterri, 2019-07-24:
### Description

[The function "Moving window peak-to-peak threshold" for epoched EEG file cannot be used. EEGLAB said that this function is part of the ERPLAB toolbox.]

<br>
#### Steps to Reproduce

1. [ERPLAB --> Artifact detection in epoched data --> Moving window peak-to-peak threshold]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[Dialogue box pops up as below.
    
    EEGLAB error
    EEGLAB error in function pop_artmwppth() at line 87:
    Index exceeds the number of array elements (1).]

<br>
#### Versions
[image]

|   |   |
|----| ---- | 
| OS version     | [Windows 10] |
| Matlab version | [2019a] |
| EEGLAB version | [2019.0] |
| ERPLAB version | [7.0] |

andrewxstewart, 2019-07-25:
Hi Psyterri,

Thanks for the report.

I would suggest trying to clear the memory of previous settings, and trying again.

You can do that with ERPLAB > Settings > ERPLAB memory settings > reset
[image]

If that doesn't resolve this, could you try emailing the EEG set you have to erplabtoobox at gmail?

psyterri, 2019-07-26:
Dear Andrew,

The function works properly now. Thank you for your help.

Best,
Terri

closed 2019-07-26 by psyterri