ucdavis/erplab #118: ERPLAB gui cannot show full pop-up window

wycalan, 2019-10-28:
### Description

I am teaching students to use erplab. But some students' computer cannot show the full pop-up window, the upper or right part of the pop-up window is missing. For example, for my computer, the right part of the pop-up window cannot show, like this:

[image]

OS version: win10, with a resolution of 1080*1920
| Matlab version | 2016B |
| EEGLAB version | 14.1.2 |
| ERPLAB version | 7.0.0 |

andrewxstewart, 2019-10-30:
Hi Wycalan,

Strange!
Does this still happen in the latest commit? Find it at:
https://github.com/lucklab/erplab/archive/master.zip

wycalan, 2019-10-30:
thanks, this problem is solved in this new version of erplab.

closed 2019-10-30 by wycalan