ucdavis/erplab #126: overlapping event latency in epoch2continuous

anomalosepia, 2020-02-14:
### Description

I don't think that this has been raised yet. 
If you are converting epoched to continuous and there are events with the same latency in the EEG.event.latency, you lose all but the first event (in the order in the latency struct) during the conversion. This is silent and doesn't raise errors or warnings. 

]

<br>
#### Steps to Reproduce

1. Get some continuous data with event structure that looks something like below. This is from EGI netstation so the data can get kind of mangled along the way. 

duration |**type**|**latency**|**epoch**|**urevent**
:-----:|:-----:|:-----:|:-----:|:-----:
600|Scrt'|99601|167|602
1|rscrt'|99685|167|603
0|'TRSP'|100193|167|604
0|'TRSP'|**_100201_**|168|605
600|'Fix'|**_100201_**|168|606
0|'bgin'|100274|168|607
0|'fix+'|100284|168|608

2. run this on your data
`EEG = epoch2continuous(EEG)`

3. the resulting structure will not have the 'Fix' event at  100201 because line 27 in epoch2continuous only takes the first event at any given latency

`[xlat, indx]  = unique_bc2([EEG.event.latency], 'first');`

in my case I was able to change that to

`[xlat, indx]  = unique_bc2([EEG.event.latency], 'last');`

but that's only for the _very_ narrow case of the data that I have here. 

<br>
#### Expected behavior:
I expected all the events to carry over from the epoched data to the continuous, with boundary event. data

<br>
#### Actual behavior:
Only the first event comes over. 

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | mac os x catalina|
| Matlab version | 2018b |
| EEGLAB version | 2019.1
| ERPLAB version | 7.0 |

andrewxstewart, 2020-02-14:
Thanks for reporting this bug with epoch2continuous.

I'll take a look at this.

anomalosepia, 2020-02-14:
Thank you! 
I just removed the requirement for unique latencies and it seems to be sorted out. 
Still checking though. 
A