ucdavis/erplab #127: ERPLAB channel operations collide with EEGLAB channel locations

leehep200286, 2020-02-25:
### When I re-reference channels using ERPLAB, it erases the channel locations in EEGLAB. Then, I am unable to perform ICA and inspect components by map. Am I doing something wrong? 

Thank you!!

(latest versions of both EEGLAB and ERPLAB, Windows 10 64bit)

ammsimmons, 2020-04-22:
This is not a fix, but you could just re-add the channel locations prior to ICA (if you have the same location file). Therefore, I usually always add the channel locations at the end of processing before ICA. ICA will run, and then you will be able to inspect component maps.

andrewxstewart, 2020-04-27:
If the properties of the channels are modified through Channel Operations, it is possible that the location information can become obsolete. This is why Channel Operations sometimes removes channel information, so as to avoid incorrect locations.

If you are just re-referencing in the Channel Operations window, you can use a tickbox in the GUI, that says "Try to preserve location info". With this selected, and the number of channels is exactly the same before and after the operation, then the location info should be kept.