ucdavis/erplab #131: data download

missyang326, 2020-05-14:
### Description

[*Description of the bug or feature*]
i can not download the datasheet,can someone send it to me ?
my mailbox:13615423782@163.com

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |
| ERPLAB version | [*fill in*] |

andrewxstewart, 2020-05-14:
It seems that the Box.com site that is used for hosting may face restrictions in some countries.

You can see other example EEG datasets, for example here:
https://osf.io/574vp/download

Does that work?

missyang326, 2020-05-14:
Thank you very much!
I have downloaded the datasets from the site that you sent to me.

https://osf.io/574vp/download