ucdavis/erplab #132: No Bins found

manukrishnamurthy, 2020-06-02:
When assigning bins from a script instead of the gui, no bins are found. 

When using the gui on a dataset, I import the dataset, import the channel locations, create an eventlist, filter the data, re-reference the data to the Cz electrode, and assign bins using a bin description file, and the process works smoothly, with the desired results. When running the same steps in a script 
`EEG = pop_loadbv("095-1", "inb_eeg_"+"095-1"+".vhdr");`
    `EEG = pop_editset(EEG, 'setname', '095-1', 'chanlocs','ERP_Analysis\brainvision_32chan_locs.ced');`
    `EEG  = pop_creabasiceventlist( EEG , 'Eventlist', strcat('095-1','\elist.txt'), 'BoundaryNumeric', { -99 }, 'BoundaryString', { 'boundary' }, 'Warning', 'on' );`
    `EEG  = pop_basicfilter( EEG,  1:32 , 'Cutoff', [ 0.3 20], 'Design', 'butter', 'Filter', 'bandpass', 'Order', 2 );`
    `pop_saveset( EEG, 'filename', s3, 'filepath', '095-1' );`
    `EEG = pop_reref(EEG, 'Cz');`
    `EEG=pop_binlister(EEG,'BDF','ERP_Analysis\Data\bin_description.txt','IndexEL',  1, 'SendEL2', 'EEG', 'Voutput', 'EEG' );`
 I get the error :
Successful Trials per bin :	0  0  0  0  0 

Full error message: 
 Bins were not found!
Try with other BDF or modify the current one.
If you think this is a bug, please report the error to erplabtoolbox@gmail.com and not to the EEGLAB developers.

binlister process was cancel.

There isn't anything wrong with my data or the program, since everything is working fine with the gui, btu how can I change the code to get it to work as well? I took this code straight from the console output when using the gui:

%Equivalent command:
EEG  = pop_binlister( EEG , 'BDF', 'ERP_Analysis\Data\bin_description.txt',...
 'IndexEL',  1, 'SendEL2', 'EEG', 'Voutput', 'EEG' ); % GUI: 02-Jun-2020 10:28:50

Thanks

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Windows 10] |
| Matlab version | [2019a] |
| EEGLAB version | [2019.0] |
| ERPLAB version | [7.0] |

andrewxstewart, 2020-06-03:
Hi there,

It could be that the bin specification in your `'ERP_Analysis\Data\bin_description.txt'` is catching zero matches in that currently loaded EEG set.

You can enter `pop_squeezevents(EEG)` to see a summary of all the events in your EEG set and the number of appearances of each. You can also view the order through `EEGLAB > Plot > Channel Data (scroll)`

The bin_description.txt does require syntax that specifies exact order. See:
https://github.com/lucklab/erplab/wiki/Assigning-Events-to-Bins-with-BINLISTER#bin-descriptor-cheat-sheet-and-examples

What does your bin_description.txt look like? Are all events present in your EEG set? And in the uninterrupted order that the bin_description.txt is looking for?

Best,
Andrew

manukrishnamurthy, 2020-06-03:
Hi My bin_description.txt file looks like 

bin 1
Auditory Non-Social
.{1}

bin2
Auditory Social
.{2}

bin 3
Visual Non-Social
.{3}

bin 4
Visual Social
.{4}

bin 5
Attention Getter
.{5}

And works fine when using the gui. All events are present in the eeg set, and although not in the same order as the bin description file, when using the same file with the gui results in bins being found and the process working properly, leading me to believe there's something wrong with the line of code specifically. What might that be?

Thanks

andrewxstewart, 2020-06-03:
Hmm - I see.

Does anything change if you provide the full path to the pop_binlister call?

Something like:
`bdf_full_path = 'C:\your_eeg_stuff\ERP_Analysis\Data\bin_description.txt';`
`EEG=pop_binlister(EEG,'BDF',bdf_full_path,'IndexEL', 1, 'SendEL2', 'EEG', 'Voutput', 'EEG' );`

manukrishnamurthy, 2020-06-03:
Unfortunately not, I tried that as well.

andrewxstewart, 2020-06-03:
If you like, could you email erplabtoolbox@gmail.com with your EEG set and that bin_description.txt file? Perhaps I can try to recreate the problem and bugfix.

sujatasinha, 2020-06-29:
Hi All, I am facing  a siilar problem. Did you find the solution?

andrewxstewart, 2020-06-29:
Hi Sujatasinha,

I don't think this is resolved yet.

If you like, could you also email erplabtoolbox@gmail.com with your EEG set and that bin_description.txt file? Perhaps I can try to recreate the problem and bugfix.

sujatasinha, 2020-06-29:
Yes, Sure!!
I am sending the EEG file, event list file and BDF.

Thanks and regards
Sujata Sinha

sujatasinha, 2020-06-30:
Hi Andrew,

I have sent the files to the email address you mentioned.

Thanks and regards
Sujata Sinha