ucdavis/erplab #136: PopUp Menu Problem

phdvictim, 2020-07-25:
### Description
I'm having a trouble with opening bigger pics in the plot erp waveforms menu. After seeing small pics I want to make these bigger but I can't click on the title. There's a popup menu blocks my cursor. How can I solve this?
[image]

[*Description of the bug or feature*]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |
| ERPLAB version | [*fill in*] |

andrewxstewart, 2020-12-03:
Hmm. That new figure overlay in Matlab is kind of distracting here.

Were you able to resolve this? I think quickly clicking once, near the electrode label, should still open up a new plot. Does that work?

If not, we may need to modify the ERP plot for many electrodes in new Matlab versions.

phdvictim, 2020-12-04:
I'm trying to click as soon as possible and it opens but after that when i try to click another one it's impossible to open that. I want to see graphs whenever I want, this problem drive me crazy.