ucdavis/erplab #137: Installing ERPLAB reordered EEGLAB menus (Glitch?)

RenzoLanfranco, 2020-08-05:
### Description

When I install ERPLAB, EEGLAB menus are reordered as shown in the image attached. I don't know how to fix this.

<br>
#### Steps to Reproduce

<br>
#### Expected behavior:
The order of menus should be 'File', 'Edit', 'Tools', 'Plot', 'Study', 'ERPLAB', 'ERPsets', 'Datasets', and 'Help'

<br>
#### Actual behavior:
The menus are ordered this way, always: 'ERPLAB', 'ERPsets', 'Datasets', 'Help', 'File', 'Edit', 'Tools', 'Plot', 'Study'.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 10*] |
| Matlab version | [*2020b*] |
| EEGLAB version | [*v2019.1*] |
| ERPLAB version | [*8.01
[image]
*] |

andrewxstewart, 2020-08-05:
Hi Renzo.

Thanks for the bug report.

Does clearing the EEGLAB/plugin/ERPLAB folder and restarting EEGLAB fix the order?

Also try restarting Matlab, and ensuring only the path to eeglab.m is added to the Matlab path - not with subfolders.

Did you install by manually placing the ERPLAB folder in EEGLAB/plugin/ERPLAB, or through the EEGLAB plugin manager?

RenzoLanfranco, 2020-08-05:
Hi Andrew,

Yes, the only thing that solves the issue is taking the ERPLAB folder out of the plugins folder (or deleting it). I installed it by manually placing the ERPLAB folder. I've been using ERPLAB for years. It's the first time that this happens.

Yes, only the eeglab folder (without subfolders is in the path.

andrewxstewart, 2020-08-05:
Huh.

What is the name of the ERPLAB plugin folder path? Is it something like:
C:\m\eeglab2019_1\plugins\erplab\eegplugin_erplab.m

Does restarting Matlab do anything?

RenzoLanfranco, 2020-08-05:
No, restarting Matlab doesn't do a thing.

Yes, the folder path is: C:\...\MATLAB\eeglab2019_1\plugins\ERPLAB8.02\eegplugin_erplab.m

andrewxstewart, 2020-08-05:
Super weird.

The only thing I can think off is other path issues.

Is there anything else in your Matlab path?

Either thru the GUI, or by entering 'PATH', check to see there is
definitely only a single EEGLAB folder, at C:...\MATLAB\eeglab2019_1

And then, thru file explorer, make sure there is only a single ERPLAB
folder in

C:...\MATLAB\eeglab2019_1\plugins

Thanks again for reporting this. Let me know if you manage to resolve it.

Best,
Andrew

RenzoLanfranco, 2020-08-05:
Yes. With the exception of eeglab folder, I erased all the other documents/MATLAB? folders from the path. It didn't make any difference.
Yes, there's only one erplab folder. I've reinstalled it many times. It's a bug that seems to come in the 8.0+ versions. I had the same issue with the 8.01 version.
EEGLAB works fine. It's only the order of the menus that is bugged.

marcusheldmann, 2020-10-05:
Hi Andrew, I observed the same issue, but only with Matlab 2020a and b. Using the identical eeglab/erplab path under 2019b the menu structure in eeglab is correct. 
Marcus

andrewxstewart, 2020-12-03:
It seems Matlab R2020a changed how the 'position' argument in uifigures was addressed.

I have a fix for that in the new release of ERPLAB, v8.10. This puts ERPLAB at the end of the EEGLAB file menu, which should be an improvement for Matlab R2020a and later.

See:
[image]

Does this work for you all?

andrewxstewart, 2020-12-03:
ERPLAB v8.10 can be found here:
https://github.com/lucklab/erplab/releases