ucdavis/erplab #140: Add ERP Lab to Open Neuroscience

amchagas, 2020-09-25:
Hello!

We are reaching out because we would love to have your project listed on [Open Neuroscience](<https://open-neuroscience.com>), and also share information about this project:

Open Neuroscience is a community run project, where we are curating and highlighting open source projects related to neurosciences!

Briefly, we have a website where short descritptions about projects are listed, with links to the projects themselves, their authors, together with images and other links. 

Once a new entry is made, we make a quick check for spam, and publish it. 

Once published, we make people aware of the new entry by [Twitter](https://twitter.com/openneurosci) and a [Facebook](https://www.facebook.com/OpenNeuroscience) group. 

**To add information about their project, developers only need to fill out this [form](https://forms.gle/ByM8thAhZJkHBMQN8)**

In the form, people can add subfields and tags to their entries, so that projects are filterable and searchable on the website!

The reason why we have the form system is that it makes it open for everyone to contribute to the website and allows developers themselves to describe their projects! 

Also, there are so many amazing projects coming out in Neurosciences that it would be impossible for us to keep track and log them all!

Open Neuroscience tech stack leverages open source tools as much as possible:
- The website is based on HUGO + Academic Theme
- Everything is hosted on github [here](https://github.com/open-neuroscience)
- We use [plausible.io](https://[plausible.io) to see visit stats on the website. It respects visitors privacy, doesn't install cookies on your computer
  - You can check our visitor stats [here](https://plausible.io/open-neuroscience.com?period=60d)

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |
| ERPLAB version | [*fill in*] |

andrewxstewart, 2020-11-18:
Thanks Andre. We would be glad to have ERPLAB listed on Open Neuroscience. I'll fill in that form now.

amchagas, 2020-11-27:
https://open-neuroscience.com/post/erplab/

closed 2020-11-27 by amchagas