ucdavis/erplab #141: Are SEM correctly plotted in ERPLAB?

RenzoLanfranco, 2020-11-09:
Hi, this is a general question. I've noticed that SEM shades in ERPLAB are always much much wider than the SEM that statistical packages give you and than the SEMs that stdshade.m gives you. I've tested it with several data sets and I've confirmed this. I was wondering if there are rules to use that option in ERPLAB when running a grand average and subsequently plotting them in ERP plotting waveforms. All the EEG researchers I've spoken to have told me that ERPLAB SEMs are wrong and that no one should use them. It's like common knowledge in the field, but I couldn't find anyone tackling this issue online. Is there a way to make ERPLAB plot the SEM correctly?

Thanks.

andrewxstewart, 2020-11-10:
Hi Renzo,

I've actually just updated the Standard Error code in ERPLAB to be more clear and more robust.

You can see the commit here:
https://github.com/lucklab/erplab/commit/21fa17f5aa36456b3d617a77d47df4a46aaa17c1

And download the beta ERPLAB using this commit here:
https://github.com/lucklab/erplab/archive/master.zip

The normal formula for sample Standard Deviation is rearranged a little, as sum_of_all_squared is already computed.

I have tested Grand Averages of example datasets, and this yields the correct numbers in the Grand Average ERP.binerror, which is what is used to plot the ERP SEM.

Does this beta release give Standard Error that works for you?

RenzoLanfranco, 2020-11-10:
Thanks so much, @andrewxstewart 
Indeed, this solves the issue. It gives the correct SEMs just like stdshade.m and stats packages do.