ucdavis/erplab #142: The contents displayed in the window were incomplete in the ERPLab

Lee666666, 2020-11-18:
### Description
Just want to know how to zoom the contents of the window to display them completely. When I run EEGLAB with my laptop and created an EventList using ERPLAB > EventList > Create EEG EventList, and then I clicked the ADVANCED button. However, I couldn't see the whole contents as shown in the figure below.
[image]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |
| ERPLAB version | [*fill in*] |

andrewxstewart, 2020-12-03:
The Advanced Eventlist GUI was about 800 pixels tall, and so could be cut off on some screens.

I've adjusted this a little in ERPLAB v8.1, to be around 680 pixels tall. It should now fit on almost all modern screens no problem.

Does this fix it for you?
https://github.com/lucklab/erplab/releases

Lee666666, 2020-12-07:
> The Advanced Eventlist GUI was about 800 pixels tall, and so could be cut off on some screens.
> 
> I've adjusted this a little in ERPLAB v8.1, to be around 680 pixels tall. It should now fit on almost all modern screens no problem.
> 
> Does this fix it for you?
> https://github.com/lucklab/erplab/releases

By reducing the scaling, I have fixed it. Thanks

closed 2020-12-10 by andrewxstewart