ucdavis/erplab #150: EEG.EVENTLIST.trialsperbin does not get updated after pop_artmwppth function deletes bins

DouweHorsthuis, 2021-09-24:
### Description

When I use pop_artmwppth to delete EEG data that is too noisy, it does not update EEG.EVENTLIST.trialsperbin. This caused me to expect way more trials. 

<br>
#### Steps to Reproduce

EEG = pop_loadset('filename', [subject_list{s} '_inter.set'], 'filepath', data_path);
EEG  = pop_creabasiceventlist( EEG , 'AlphanumericCleaning', 'on', 'BoundaryNumeric', { -99 }, 'BoundaryString', { 'boundary' } ); 
EEG  = pop_binlister( EEG , 'BDF', [binlist_location '\binlist_nback.txt'], 'IndexEL',  1, 'SendEL2', 'EEG', 'Voutput', 'EEG' ); 
EEG = eeg_checkset( EEG );
EEG = pop_epochbin( EEG , epoch_time,  baseline_time); %epoch size and baseline size
EEG= pop_artmwppth( EEG , 'Channel', 1:EEG.nbchan, 'Flag',  1, 'Threshold',  120, 'Twindow', epoch_time, 'Windowsize',  200, 'Windowstep',  200 );
EEG = pop_rejepoch( EEG, [EEG.reject.rejmanual] ,0);

<br>
#### Expected behavior:
[1- Excpected flagged data to be deleted
[2- Excpected EEG.EVENTLIST.trialsperbin to decrease after the running pop_rejepoch

<br>
#### Actual behavior:
[Data was deleted, but EEG.EVENTLIST.trialsperbin is not updated]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Windows 10] |
| Matlab version | [2019b] |
| EEGLAB version | [2021_1] |
| ERPLAB version | [8.10] |

DouweHorsthuis, 2021-09-24:
solution could be similar to what @andrewxstewart suggested in issue #103 .
But it's not clear to me how to do that

ammsimmons, 2021-09-28:
Hi. The artifact detection routines in ERPLAB do not reject trials/epochs. They only mark/flag problematic epochs. The epochs still remain in your dataset. Rejection occurs during averaging. That is, epochs marked for rejection are not included in the ERP averages (and the number of trials of the ERPs reflect this exclusion of marked epochs). 

The artifact detection routines do update the EVENTLIST structure  (in the EEG.reject.rejmanual and EEG.reject.rejmanualE fields), so it possible to query these fields and obtain information about the trials that passed the artifact detection routine. Please read more here: https://github.com/lucklab/erplab/wiki/Artifact-Detection-in-Epoched-Data