ucdavis/erplab #157: Flagged EEG Portions not Shown after "pop_continuousartdet()"

samuelsze333, 2022-06-21:
Hello everyone. I would like to ask for help regarding viewing flagged EEG data after applying "pop_continuousartdet()" with a script.

### Description
When I run "pop_continuousartdet()" on the GUI, a window showing flagged EEG data portions pops up, just like the one below:
[image]

However, if I run "pop_continuousartdet()" with exactly the same parameters, the window with flagged EEG data does not pop up. When I plot the data, the flagged portions have already been removed, just like below:
[image]

<br>
#### Steps to Reproduce
I run the script below, and then the problem described occurs. The final line is where I apply "pop_continuousartdet()".

[ALLEEG EEG CURRENTSET ALLCOM] = eeglab;

EEG = pop_loadcnt('C:\Users\85296\Desktop\EEG Datasets\Acquisition 782_2019003 Data.cnt' , 'dataformat', 'auto', 'memmapfile', '');

[ALLEEG EEG CURRENTSET] = pop_newset(ALLEEG, EEG, 0,'setname','CNT file','gui','off'); 

eeglab redraw;

EEG = eeg_checkset( EEG );

EEG = pop_saveset( EEG, 'filename','CNT782.set','filepath','C:\\Users\\85296\\Desktop\\EEG Datasets\\');
eeglab redraw;

[ALLEEG EEG] = eeg_store(ALLEEG, EEG, CURRENTSET);

EEG  = pop_basicfilter( EEG,  1:34 , 'Boundary', 'boundary', 'Cutoff',  30, 'Design', 'butter', 'Filter', 'lowpass', 'Order',  2 ); % GUI: 12-Jun-2022 01:04:24

[ALLEEG EEG CURRENTSET] = pop_newset(ALLEEG, EEG, 1,'gui','off'); 
eeglab redraw;

EEG  = pop_basicfilter( EEG,  1:34 , 'Boundary', 'boundary', 'Cutoff',  0.1, 'Design', 'butter', 'Filter', 'highpass', 'Order',  2 ); % GUI: 12-Jun-2022 01:04:43

[ALLEEG EEG CURRENTSET] = pop_newset(ALLEEG, EEG, 2,'gui','off'); 
eeglab redraw;

EEG = pop_cleanline(EEG, 'bandwidth',2,'chanlist',[1:34] ,'computepower',1,'linefreqs',50,'newversion',0,'normSpectrum',0,'p',0.01,'pad',2,'plotfigures',0,'scanforlines',0,'sigtype','Channels','taperbandwidth',2,'tau',100,'verb',1,'winsize',4,'winstep',1);

[ALLEEG EEG CURRENTSET] = pop_newset(ALLEEG, EEG, 3,'gui','off'); 
eeglab redraw;

EEG = eeg_checkset( EEG );

EEG=pop_chanedit(EEG, 'lookup','C:\\Users\\85296\\Desktop\\eeglab2022.0\\plugins\\dipfit\\standard_BEM\\elec\\standard_1005.elc');

[ALLEEG EEG] = eeg_store(ALLEEG, EEG, CURRENTSET);
eeglab redraw;

EEG = eeg_checkset( EEG );

EEG=pop_chanedit(EEG, 'append',34,'changefield',{35,'labels','CPz'},'lookup','C:\\Users\\85296\\Desktop\\eeglab2022.0\\plugins\\dipfit\\standard_BEM\\elec\\standard_1005.elc','setref',{'1:33','CPz'});

[ALLEEG EEG] = eeg_store(ALLEEG, EEG, CURRENTSET);
eeglab redraw;

EEG  = pop_basicfilter( EEG,  1:34 , 'Boundary', 'boundary', 'Cutoff',  30, 'Design', 'butter', 'Filter', 'lowpass', 'Order',  8 ); % GUI: 12-Jun-2022 01:14:52

[ALLEEG EEG CURRENTSET] = pop_newset(ALLEEG, EEG, 4,'gui','off'); 
eeglab redraw;

EEG  = pop_basicfilter( EEG,  1:34 , 'Boundary', 'boundary', 'Cutoff',  0.1, 'Design', 'butter', 'Filter', 'highpass', 'Order',  8 ); % GUI: 12-Jun-2022 01:15:10

[ALLEEG EEG CURRENTSET] = pop_newset(ALLEEG, EEG, 5,'gui','off');
eeglab redraw;

EEG = eeg_checkset( EEG );

pop_eegplot( EEG, 1, 1, 1);

EEG = eeg_eegrej( EEG, [87519 96164;104528 146344;404157 416006;674467 749300;1015247 1032365;1289973 1295093;1545985 1626512;1894025 1913666;2174042 2221980;2476050 2513770;2773975 2806492;3073980 3106623;3387420 3407927;3669015 3685517;3685530 3691457;3953953 3971994;4063369 4067500]);

[ALLEEG EEG CURRENTSET] = pop_newset(ALLEEG, EEG, 6,'gui','off'); 

EEG = pop_continuousartdet( EEG , 'ampth',  500, 'chanArray',  1:31, 'colorseg', [ 1 0.9765 0.5294], 'forder',  100, 'numChanThreshold',  1, 'stepms',  250, 'threshType', 'peak-to-peak', 'winms',  1000 ); % GUI: 12-Jun-2022 01:41:49

<br>
#### Expected behavior:
The window showing flagged data portions pops up.

<br>
#### Actual behavior:
The window showing flagged data portions does not pop up.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Window 11] |
| Matlab version | [R2022A] |
| EEGLAB version | [2022.0] |
| ERPLAB version | [9.00] |

Thank you so much for your help!!!

andrewxstewart, 2022-06-21:
https://github.com/lucklab/erplab/issues/157

Hi Samuelsze,

Thanks for the report.

I think pop_continuousartdet works both when run from the GUI, and when the function is called outside the GUI.

By default, at the moment, the EEGPLOT summary window is only shown when this is run from the GUI.

This avoids loads of windows being opened when a script is run on many subjects.

You can still choose to show EEGPLOT in a script with the optional argument:
EEG = pop_continuousartdet(EEG, 'review', 'on')

You can see a little more info with:
help pop_continuousartdet

Or:
edit pop_continuousartdet
then look around line 437 for:
%% Added in option for skipping EEGPLOT GUI when scripting

Does that work for you?

Best,
Andrew Stewart

samuelsze333, 2022-06-21:
Dear Mr. Stewart,

I am so grateful for your reply and help. I have tried including the optional argument with the keyword " 'review' " and the value " 'on' ", and a window does pop up. Nevertheless, the window does not show flagged portions but continuous EEG data with all flagged portions removed. Please see below:
[image]

The code I used is: 
EEG = pop_continuousartdet( EEG , 'ampth',  500, 'chanArray',  1:31, 'colorseg', [ 1 0.9765 0.5294], 'forder',  100, 'numChanThreshold',  1, 'stepms',  250, 'threshType', 'peak-to-peak', 'winms',  1000, 'review', 'on' );

Could I ask whether you could offer me some advice on this? Thank you so much, Mr. Stewart.

samuelsze333, 2022-06-22:
> #157
>
> Hi Samuelsze,
>
> Thanks for the report.
>
> I think pop_continuousartdet works both when run from the GUI, and when the function is called outside the GUI.
>
> By default, at the moment, the EEGPLOT [...]

Dear Mr Stewart,

I restarted EEGLAB and followed your instruction again, and that window pops up. Thank you so much for your help!!!

Have a good one.

Best regards,
Samuel

andrewxstewart, 2022-06-22:
Glad this was useful.

Perhaps we should make this option more salient in ERPLAB.
@ammsimmons @guanghuizhang0328 ?

closed 2022-06-22 by andrewxstewart