ucdavis/erplab #159: Error when starting EEGlab

PaulSteinfath, 2022-07-22:
### Description

When I start eeglab I get the following error:

An error occurred while initializing child process: While attempting to execute "mv -f
"/data/pt_02584/Patty/Toolboxes/eeglab2021.1/plugins/ERPLAB9.00/erplab_Box/eventlist_backup_7387245778703704"
"/tmp/MATLAB_Files_22-Jul-2022_13_30_12/eventlist_backup_7387245778703704"" in context { environment : { ARCH : glnxa64,
AUTOMOUNT_MAP : , BASEMATLABPATH : , DBUS_SESSION_BUS_ADDRESS : unix:path=/run/user/13438/bus, DISPLAY : localhost:17.0, ENVLIST :
matlab=9.12, FB_COLORIZE : 0, GFORTRAN_STDERR_UNIT : -1, GFORTRAN_STDIN_UNIT : -1, GFORTRAN_STDOUT_UNIT : -1, HISTCONTROL :
ignoreboth, HISTSIZE : 50000, HOME : /data/hu_steinfath, ICU_TIMEZONE_FILES_DIR :
/afs/cbs.mpg.de/software/matlab/9.12/bin/icutzdata, KMP_BLOCKTIME : 1, KMP_HANDLE_SIGNALS : 0, KMP_INIT_AT_FORK : false,
KMP_STACKSIZE : 512k, KRB5CCNAME : FILE:/tmp/krb5cc_13438_zXLSeNr1ak, LANG : en_US.UTF-8, LANGUAGE : en_US:en, LC_NUMERIC : C,
LD_LIBRARY_PATH :
/afs/cbs.mpg.de/software/matlab/9.12/sys/os/glnxa64:/afs/cbs.mpg.de/software/matlab/9.12/bin/glnxa64:/afs/cbs.mpg.de/software/matlab/9.12/extern/lib/glnxa64:/afs/cbs.mpg.de/software/matlab/9.12/cefclient/sys/os/glnxa64:/afs/cbs.mpg.de/software/matlab/9.12/runtime/glnxa64:/afs/cbs.mpg.de/software/matlab/9.12/sys/java/jre/glnxa64/jre/lib/amd64/native_threads:/afs/cbs.mpg.de/software/matlab/9.12/sys/java/jre/glnxa64/jre/lib/amd64/server,
LD_PRELOAD : , LOCALPROFILE : cbs-fb, LOGNAME : steinfath, MATLAB : /afs/cbs.mpg.de/software/matlab/9.12, MATLABPATH : , MCRROOT :
/afs/cbs.mpg.de/software/matlab/9.12/debian-bullseye-amd64, MEMKIND_HEAP_MANAGER : TBB, MKL_DOMAIN_NUM_THREADS : , MKL_NUM_THREADS
: , MOTD_SHOWN : pam, OLDPWD : /afs/cbs.mpg.de/software/matlab/9.12, OSG_LD_LIBRARY_PATH :
/afs/cbs.mpg.de/software/matlab/9.12/sys/openscenegraph/lib/glnxa64, PAGER : /usr/bin/less, PARALLEL_SERVER_EXPERIMENTAL_MJS_VER :
, PARALLEL_SERVER_EXPERIMENTAL_PLR_VER : , PARALLEL_SERVER_EXPERIMENTAL_SER_TYPE : , PATH :
/afs/cbs.mpg.de/software/matlab/9.12/debian-bullseye-amd64/bin:/bin:/sbin:/opt/bin:/usr/bin:/usr/sbin:/afs/cbs.mpg.de/software/scripts:/afs/cbs.mpg.de/software/scripts/science:/afs/cbs.mpg.de/software/bin/linux:/usr/games,
PERL5LIB : /afs/cbs.mpg.de/software/perllib, PRE_LD_PRELOAD : , PWD : /data/hu_steinfath, PYTHONPATH :
/afs/cbs.mpg.de/software/pythonlib/python3.5:/afs/cbs.mpg.de/software/matlab/9.12/debian-bullseye-amd64/extern/engines/python/dist,
SHELL : /bin/bash, SHINTERACTIVE : 1, SHLVL : 0, SHTERM : 1, SSH_CLIENT : 10.0.65.128 39872 22, SSH_CONNECTION : 10.0.65.128 39872
10.0.68.147 22, SSH_TTY : /dev/pts/9, TERM : xterm-256color, TOOLBOX : /afs/cbs.mpg.de/software/matlab/9.12/toolbox, USER :
steinfath, XDG_RUNTIME_DIR : /run/user/13438, XDG_SESSION_CLASS : user, XDG_SESSION_ID : 1795, XDG_SESSION_TYPE : tty,
XFILESEARCHPATH : /afs/cbs.mpg.de/software/matlab/9.12/sys/java/jre/glnxa64/jre/lib/locale/%L/%T/%N%S:, __KMP_REGISTERED_LIB_82474
: 0x7fca99531708-cafef591-libiomp5.so }, std_in : inherit_from_parent, std_out : pipe_to pipe { source : 931, sink : 932 },
std_err : merge_with_stdout, directory : unspecified, user : unspecified, inherited_handles : {  }, detached_process : --,
explicitly inherit handles : --, create_no_window : -- }: No such file or directory (trace: execve
<fork>
)

<br>

The error is temporarily solved if I clear all the files in:
 /data/pt_02584/Patty/Toolboxes/eeglab2021.1/plugins/ERPLAB9.00/erplab_Box/

However, when I run some analyses the folder "erplab_Box" fills up and I get the same error again.

Does anyone have advice on how to resolve this?

<br>
#### Versions
|   |   |
|----| ---- | 
| OS version     | Debian GNU/Linux 11 (bullseye) |
| Matlab version | 2202a|
| EEGLAB version | 2021.1 / 2022 |
| ERPLAB version | 9|

andrewxstewart, 2022-07-22:
Hi Paul. Does Matlab have full write access to the folder in which EEGLAB + ERPLAB is installed?

Do you get the same error if you try copying the EEGLAB + ERPLAB folder from the current path at /data/pt_02584/Patty/Toolboxes/eeglab2021.1 to instead be on something like /home/toolboxes/eeglab2021.1 ?

PaulSteinfath, 2022-07-25:
Hi Andrew,

Thank you for the advice - I tried different folders, but still get the error:

 "An error occurred while initializing child process: While attempting to execute "mv -f
"/data/hu_steinfath/Toolbox_temp/eeglab2021.1/plugins/ERPLAB8.10/erplab_Box/eventlist_backup_7387274287962962"
"/tmp/MATLAB_Files_25-Jul-2022_09_55_57/eventlist_backup_7387274287962962"" in context { environment : { ARCH : glnxa64, AUTOMOUNT_MAP : , BASEMATLABPATH : ,
DBUS_SESSION_BUS_ADDRESS : unix:path=/run/user/13438/bus, DISPLAY : localhost:13.0, ENVLIST : matlab=9.12, FB_COLORIZE : 0, GFORTRAN_STDERR_UNIT : -1,
GFORTRAN_STDIN_UNIT : -1, GFORTRAN_STDOUT_UNIT : -1, HISTCONTROL : ignoreboth, HISTSIZE : 50000, HOME : /data/hu_steinfath, ICU_TIMEZONE_FILES_DIR :
/afs/cbs.mpg.de/software/matlab/9.12/bin/icutzdata, KMP_BLOCKTIME : 1, KMP_HANDLE_SIGNALS : 0, KMP_INIT_AT_FORK : false, KMP_STACKSIZE : 512k, KRB5CCNAME :
FILE:/tmp/krb5cc_13438_FMKQl6lWzO, LANG : en_US.UTF-8, LANGUAGE : en_US:en, LC_NUMERIC : C, LD_LIBRARY_PATH :
/afs/cbs.mpg.de/software/matlab/9.12/sys/os/glnxa64:/afs/cbs.mpg.de/software/matlab/9.12/bin/glnxa64:/afs/cbs.mpg.de/software/matlab/9.12/extern/lib/glnxa64:/afs/cbs.mpg.de/software/matlab/9.12/cefclient/sys/os/glnxa64:/afs/cbs.mpg.de/software/matlab/9.12/runtime/glnxa64:/afs/cbs.mpg.de/software/matlab/9.12/sys/java/jre/glnxa64/jre/lib/amd64/native_threads:/afs/cbs.mpg.de/software/matlab/9.12/sys/java/jre/glnxa64/jre/lib/amd64/server,
LD_PRELOAD : , LOCALPROFILE : cbs-fb, LOGNAME : steinfath, MATLAB : /afs/cbs.mpg.de/software/matlab/9.12, MATLABPATH : , MCRROOT :
/afs/cbs.mpg.de/software/matlab/9.12/debian-bullseye-amd64, MEMKIND_HEAP_MANAGER : TBB, MKL_DOMAIN_NUM_THREADS : , MKL_NUM_THREADS : , MOTD_SHOWN : pam, OLDPWD :
/afs/cbs.mpg.de/software/matlab/9.12, OSG_LD_LIBRARY_PATH : /afs/cbs.mpg.de/software/matlab/9.12/sys/openscenegraph/lib/glnxa64, PAGER : /usr/bin/less,
PARALLEL_SERVER_EXPERIMENTAL_MJS_VER : , PARALLEL_SERVER_EXPERIMENTAL_PLR_VER : , PARALLEL_SERVER_EXPERIMENTAL_SER_TYPE : , PATH :
/afs/cbs.mpg.de/software/matlab/9.12/debian-bullseye-amd64/bin:/bin:/sbin:/opt/bin:/usr/bin:/usr/sbin:/afs/cbs.mpg.de/software/scripts:/afs/cbs.mpg.de/software/scripts/science:/afs/cbs.mpg.de/software/bin/linux:/usr/games,
PERL5LIB : /afs/cbs.mpg.de/software/perllib, PRE_LD_PRELOAD : , PWD : /data/hu_steinfath, PYTHONPATH :
/afs/cbs.mpg.de/software/pythonlib/python3.5:/afs/cbs.mpg.de/software/matlab/9.12/debian-bullseye-amd64/extern/engines/python/dist, SHELL : /bin/bash, SHINTERACTIVE :
1, SHLVL : 0, SHTERM : 1, SSH_CLIENT : 10.0.65.128 36908 22, SSH_CONNECTION : 10.0.65.128 36908 10.0.68.148 22, SSH_TTY : /dev/pts/5, TERM : xterm-256color, TOOLBOX :
/afs/cbs.mpg.de/software/matlab/9.12/toolbox, USER : steinfath, XDG_RUNTIME_DIR : /run/user/13438, XDG_SESSION_CLASS : user, XDG_SESSION_ID : 2474, XDG_SESSION_TYPE :
tty, XFILESEARCHPATH : /afs/cbs.mpg.de/software/matlab/9.12/sys/java/jre/glnxa64/jre/lib/locale/%L/%T/%N%S:, __KMP_REGISTERED_LIB_192501 :
0x7fe0c24d8708-cafe7686-libiomp5.so }, std_in : inherit_from_parent, std_out : pipe_to pipe { source : 945, sink : 946 }, std_err : merge_with_stdout, directory :
unspecified, user : unspecified, inherited_handles : {  }, detached_process : --, explicitly inherit handles : --, create_no_window : -- }: No such file or directory
(trace: execve
<fork>
)
 "

However, I think the problem stems from Matlab and not EEG/ERPLAB since the error occurs also when running other scripts that have nothing to do with EEG analysis. 

I still would be very happy about advice on what is going on, but guess the issue could be closed here.

Best, Paul

closed 2022-08-09 by PaulSteinfath