ucdavis/erplab #160: pop_gaverager() leads to Unrecognized function or variable 'DQ_spec_structure'

samuelsze333, 2022-09-02:
After I run Average Across ERP sets (Grand Average), the command window of MATLAB shows the code below:
ERP = pop_gaverager( ALLERP , 'DQ_flag', 1, 'DQ_spec', DQ_spec_structure, 'Erpsets', [1  2], 'ExcludeNullBin', on, 'SEM', on );

When I run the same code above as part of a script, it returns the error message below:
Unrecognized function or variable 'DQ_spec_structure'.

It is strange because when I use the GUI, ERPLAB works fine. But I when run it in a script, it does not work.

Thank you so much your help in advance.

andrewxstewart, 2022-09-02:
Hi there,

Thanks for the bug report.

DQ_spec_structure is a the Matlab structure that specifies how to construct the data quality measures.

If DQ_flag is 0, then there is no data quality measures, and so this can be empty.

If DQ_flag is 1, and the code is run from the GUI, then this DQ_spec_structure will be set up from what the user selects, or fall back to defaults.

If DQ_flag is 1, and the code is run from script, then it is expected that the user will create a correctly formatted DQ_spec_structure in the workspace before running that line....  but there also isn't great error messages or help if this is missed. We should fix this.

First, let's get this working for you.

Does it work if you add these lines to your script before running pop_averager??

`% Set up default struct for DQ_spec_structure  
DQ_spec_structure.measures = [1, 2, 3]; % Use first 3 DQ measures
DQ_spec_structure.methods = [2, 2, 2]; % Use the 2nd combo method, Root-Mean Square, for each
DQ_spec_structure.measure_names = {'Baseline Measure - SD';'Point-wise SEM'; 'aSME'};
DQ_spec_structure.method_names = {'Pool ERPSETs, GrandAvg mean','Pool ERPSETs, GrandAvg RMS'};
DQ_spec_structure.str = {'Baseline Measure - SD, GrandAvg RMS';'Point-wise SEM, GrandAvg RMS'; 'aSME GrandAvg RMS'};

ERP = pop_gaverager( ALLERP , 'DQ_flag', 1, 'DQ_spec', DQ_spec_structure, 'Erpsets', [1 2], 'ExcludeNullBin', on, 'SEM', on );`

samuelsze333, 2022-09-02:
> Hi there,
>
> Thanks for the bug report.
>
> DQ_spec_structure is a the Matlab structure that specifies how to construct the data quality measures.
>
> If DQ_flag is 0, then there is no data quality [...]

Thank you so much for your help, Mr Stewart!

I just tried it. Now the error message " Unrecognized function or variable 'DQ_spec_structure' " has gone. However, I get the error message below:

Unrecognized function or variable 'on'.

Could you please help me with that? Thank you!

andrewxstewart, 2022-09-02:
Ah - I think those should be 'on'. Try this:

```
ERP = pop_gaverager( ALLERP , 'DQ_flag', 1, 'DQ_spec', DQ_spec_structure, 'Erpsets', [1 2], 'ExcludeNullBin', 'on', 'SEM', 'on' );
```

samuelsze333, 2022-09-02:
> Ah - I think those should be 'on'. Try this:
> 
> ```
> ERP = pop_gaverager( ALLERP , 'DQ_flag', 1, 'DQ_spec', DQ_spec_structure, 'Erpsets', [1 2], 'ExcludeNullBin', 'on', 'SEM', 'on' );
> ```

I am very appreciative of your help. Your code works well!

ammsimmons, 2022-09-02:
Thanks @andrewxstewart ! It appears this is solved for now. Future code will address script compatibility.

closed 2022-09-02 by ammsimmons