ucdavis/erplab #176: Inquiry: How to combine

atiy90, 2023-10-04:
Hello there,

I am currently trying to combine ERP waveforms (congruence and incongruence stimuli) of the control and target groups into one single graph using ERPLAB. But no such option can be found in ERPLAB.

May I know now how to do it using ERPLAB? I have no scripting or coding experience. So, any guide on using only GUIs in ERPLAB would be very much appreciated. Thank you.

stevenjluck, 2023-10-04:
You can find the answer near the bottom of our FAQ page: https://github.com/ucdavis/erplab/wiki/Troubleshooting-and-Frequently-Asked-Questions#plotting-or-combining-erps-that-are-in-different-erpsets

I hope this helps!

Steve

atiy90, 2023-10-04:
I did follow the instruction and it's working now, Thank you Steve!