ucdavis/erplab #182: subtraction and comparison of amplitudes from two test conditions

mhershkow, 2023-12-09:
Hi all!

I'm doing P300 experiment. The subject is presented with different kinds of signals (e.g.: target, novel, standart).
[image]

In order to compare graphically between the different condition - I'm seeking a way to subtract the amplitudes of different condition (e.g.: Subtract the values ​​obtained in the Target condition from the values ​​obtained in Novel condition and plotting another graph with only one line presented.

Is there a way of doing it using ERPLAB?

Thank you!

stevenjluck, 2023-12-09:
Hi. I recommend doing the ERPLAB Tutorial (available from https://github.com/ucdavis/erplab/wiki). That will show you how to do things such as making difference waves.

Steve

--------------------------------------------------------------------
Steve Luck, Ph.D. (he/him/his)
Distinguished Professor, Department of Psychology<http://psychology.ucdavis.edu/>
Core Faculty, Center for Mind & Brain<https://mindbrain.ucdavis.edu>
University of California, Davis
267 Cousteau Place [directions<https://mindbrain.ucdavis.edu/directions>]
Room 126
Davis, CA 95618
(530) 754-4524
***@***.******@***.***>
http://mindbrain.ucdavis.edu/people/sjluck
http://lucklab.ucdavis.edu
http://erpinfo.org
--------------------------------------------------------------------

733.png (view on web)<https://github.com/ucdavis/erplab/assets/142025728/6ece2d88-6272-4c3c-b22e-71bcdf786b92>

mhershkow, 2023-12-09:
Dear steve,

Thank you so much for the help! I've found the answer in the ERPLAB Tutorial!

Sincerely,

Moshe