ucdavis/erplab #184: inactive ERPLAB functions  and  hidden EEGLAB menu items.

pratibhabhise01, 2024-01-10:
i use eeglab 2023.1 version but in that some menus are hidden.herewith sending the screen shot. i al so tried preference setting.
[image]
[image]

andrewxstewart, 2024-01-10:
Hi Prati,

ERPLAB has many functions that work only on ERP datasets. When there is no
ERP set loaded, those options are grayed out.

To run those, load an EEG dataset, and create a bin-averaged ERP dataset
from this.

See:
https://github.com/ucdavis/erplab/wiki/ERPLAB-Tutorial

Best,
Andrew Stewart

pratibhabhise01, 2024-01-13:
could anyone explain how to interpret the channel properties..... continuous data values .is their and values scale like if values more that 1 or greater than 1 or 2 it is good or not