ucdavis/erplab #189: Question about BinLister

thakralp, 2024-03-10:
I had a question regarding binning my data. I have 255 events, that each have a 1 of 3 event codes: 2, 3, or 4

I know based on the the response data which events of the 255 correspond to which of the 8 bins I want to segment my data into (e.g., item #'s 1, 13, 100, and 202, fall into bin 1). The issue I am having that those items have overlapping ecodes.

Is there a way to use the item# as a way to create a BDF/BinLister file that segments my bins? I have only found BDF templates that use ecodes to assign bins

Thanks!

davidg622, 2024-03-11:
Hello,

There is unfortunately no way to directly create a Binlister file that uses item#. You could, however, try to manually add/overwrite eventcodes by creating and exporting the eventlist, editing the .txt file, and importing it back into your data. This would be time consuming, but it's the only way I think you could accomplish your goal.

Hope this helps,

David

Allexxann, 2024-06-14:
The response data isn't present in your recording itself, correct?

stevenjluck, 2024-06-14:
Our example data include event codes for the responses.

Steve

Allexxann, 2024-08-06:
You haven't attached the example data, so I can't have a look, but if you're using the response data to segment the item #'s, why not just use it in the BDF file?