ucdavis/erplab #192: [ERPLAB studio] Issue with Automatic Selection of Periods to Delete in Reject Artifactual Time Segments

coolmintmild, 2024-06-26:
### Description

I'm new to ERPLAB and currently following a tutorial to familiarize myself with ERPLAB Studio. I encountered an issue with the "Reject Artifactual Time Segments" panel, specifically under the "Automatic Selection of Periods to Delete" option. It doesn't appear to be functioning as expected.

During the preview, I can see the segments marked as artifactual, and the MATLAB command window confirms this with the message "5 segments were marked." The MATLAB command used is as follows:

 5 segments were marked.

Done.  What's next?

EEG = pop_continuousartdet( EEG , 'ampth',  200, 'chanArray', [ 2:14 16:28], 'colorseg', [ 0.83 0.82 0.79], 'forder',  100, 'numChanThreshold',  1, 'shortisi',  1000, 'stepms',  500, 'threshType', 'peak-to-peak', 'winms',  1000 );
Creating a new ALLEEG dataset 1

However, after finalizing the changes, no alterations are evident in the EEG data. The segments that were marked do not show any signs of deletion or modification, and the 96-second segment highlighted in the tutorial still remains unchanged. Could someone please help me understand why the marked segments are not being deleted or altered as expected?

<br>
#### Steps to Reproduce

1. Open ERPLAB Studio and navigate to the "Reject Artifactual Time Segments" panel.
2. Preview the marked segments in the EEG data.
3. Finalize changes and review the EEG dataset.

<br>
#### Expected behavior:
I expected the marked artifactual segments to be deleted from the EEG dataset, reflecting changes when reviewed post-finalization.

<br>
#### Actual behavior:
The preview shows the segments as marked. However, after finalizing, no changes appear in the EEG dataset. The segments remain unchanged, and the 96-second segment from the tutorial is still present.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows11*] |
| Matlab version | [*R2023B*] |
| EEGLAB version | [*2024.0*] |
| ERPLAB version | [*11.01*] |

stevenjluck, 2024-06-26:
Thanks for reporting this bug (and providing such a detailed description!).

We actually discovered this bug earlier today, and we have a fix for it.

You can download the updated version by going to https://github.com/ucdavis/erplab, clicking on the green Code button, and the clicking Download ZIP (see screenshot below).

You can then move the folder into the Plugins folder inside your EEGLAB folder. You’ll need to name it ERPLAB11.0 (and delete the previous ERPLAB11.0 folder that should be there).

I hope this helps!

Steve

***@***.***

coolmintmild, 2024-06-27:
Thank you for the prompt response and for providing the updated version. I appreciate the clear instructions for downloading and installing the new version of ERPLAB.

I followed the steps as outlined, but unfortunately, I'm still experiencing the same issue where the marked artifactual segments are not being deleted post-finalization, just as before.

Could there be another step I need to ensure is configured correctly, or might there be a specific setting in the new version that requires adjustment? Any further guidance you can provide would be greatly appreciated.

coolmintmild, 2024-06-27:
I've encountered another issue with the Plot Settings panel when using a custom grid layout. After importing a custom grid layout file (GridLocations_Custom.tsv), the plot updates to reflect the new layout positions—column, row, and graph arrangement change as specified. However, despite these visual updates, the channel names remain in their original numerical order and do not update to match the new layout. I appreciate any help you can provide on this issue.

[image]

Thank you!

stevenjluck, 2024-06-28:
We are about to release an update (v11.02), which should solve both of your problems.

If it doesn’t solve them, then please send us screenshots of the relevant panels and sample data files so that we can replicate the problem.

Steve

bug1_custom-montage.png (view on web)<https://github.com/ucdavis/erplab/assets/58200092/d5566f41-e3a3-4a6a-b88e-5f915acfb6f9>