ucdavis/erplab #193: [ERPLAB Studio] Issue with non-existent fields

aymeric-collart, 2024-07-22:
### Description

Dear all, I've been trying ERPLAB Studio recently, and as I can launch the interface and load data, I don't have access to many options, including 'EEG_autoplot' as well as many others in the menu. I've tried with the original interface (EEGLAB + ERPLAB), there is no problem.

Several error messages occurred as I started ERPLAB studio, but I'm not sure how these are related to my problem:

Error: File: f_ERP_simulation_panel.m Line: 154 Column: 25
"srateop" was previously used as a variable, conflicting with its use here as the name of a function or command.
See "How MATLAB Recognizes Command Syntax" in the MATLAB documentation for details.

Error in EStudio_ERP_Tab (line 75)
EStudio_gui_erp_totl.panel{19} =
f_ERP_simulation_panel(EStudio_gui_erp_totl.settingLayout,EStudio_gui_erp_totl.panel_fonts);

Error in estudio/createInterface (line 377)
        EStudio_gui_erp_totl = EStudio_ERP_Tab(EStudio_gui_erp_totl,ColorB_def);

Error in estudio (line 200)
EStudio_gui_erp_totl = createInterface();

Here are the error messages when I load data using the GUI and when I try to click on other functions:

Reference to non-existent field 'EEG_autoplot'.

Error in f_EEG_eeg_sets_GUI/load (line 773)
        if EStudio_gui_erp_totl.EEG_autoplot==1
 
Error while evaluating UIControl Callback.

Reference to non-existent field 'eegpanel'.

Error in EStudio_EEG_Tab>nMinimize (line 160)
minned = EStudio_gui_erp_totl.eegpanel{whichpanel}.IsMinimized;
 
Error while evaluating UIControl Callback.

<br>
#### Steps to Reproduce

1. Open ERPLAB Studio (first error message)
2. Load data (using either 'Load' or 'Import')

<br>
#### Expected behavior:
EEG data plotted on the GUI
Access to other functions

<br>
#### Actual behavior:
Nothing plotted
No access to other functions (filter, edit channel locations, etc.)

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Windows 11] |
| Matlab version | [Matlab R2019a] |
| EEGLAB version | [2024.0] |
| ERPLAB version | [11.03] |

davidg622, 2024-07-22:
Hello,

Thank you for your bug report!

We have fixed the error related to the function "srateop". You can download the updated version by going to https://github.com/ucdavis/erplab, clicking on the green Code button, and the clicking Download ZIP.

You can then move the folder into the Plugins folder inside your EEGLAB folder. You’ll need to name it ERPLAB11.03 (and delete the previous ERPLAB11.03 folder that should be there).

If EStudio still doesn't work, I suspect it might be because you are using an older version of MATLAB. We currently recommend using MATLAB 2021a or sooner (see https://github.com/ucdavis/erplab/wiki/Installation for more details).

Thanks,

David

aymeric-collart, 2024-07-22:
I just gave a try, this works perfectly with the updated version. Thank you very much!

closed 2024-07-22 by aymeric-collart