ucdavis/erplab #194: Missing Event Info after loading a second file

andyleynes, 2024-07-22:
I am using the Estudio version 11.03 (latest version of EEGLab) to reprocess some data. I am loading continuous EEG files that have been resampled, filtered, and event info has been added. I then create epochs, perform ICA, check for artifacts, form ERPs etc. 

When I am ready to process the next subject's data, I clear each EEG data set from Estudio. I then load the next ".set" file and the event information is not loaded. When I look in "EEG and Bin Info" the "set name" and "file name" fields are blank and the event info is blank. If I close Estudio, restart it, and reload the data the event info is there. Something is not working properly. The previous version of Estudio (11.02?) did not have the same issue. 

Separate request for the future: it would be nice to have a "clear all" in the EEG window rather than having to clear multiple open data sets.

davidg622, 2024-07-22:
Hello,

Thank you for bringing this issue to our attention. Unfortunately, I haven't been able to reproduce the problem on my end. Could you please provide more details about your setup? Specifically, I'd like to know your MATLAB version and the operating system you're using.

Additionally, just to clarify, you can clear multiple .set files by selecting several of them before pressing "clear."

Best regards,
David

andyleynes, 2024-07-22:
Hi David,
Matlab version R2021a, eeglab 2024.0, windows 10.2205 x64.

Thanks for the tip about clearing files. I tried that (maybe ver 11.02?)
and it didn't work. But it does now! And I don't experience the missing
event info when I clear multiple EEG files in one click but it does happen
when you press 'clear' multiple times (I have 8 files open at the end).
That should narrow the problem.

davidg622, 2024-07-22:
Hello,

I attempted to recreate the issue using the "clear" function as you described on both my macOS and Windows 10 systems. Unfortunately, I was unable to replicate the problem. The next step would be for you to send us some sample data so we can better isolate the issue. Please email it to me at drgarrett@ucdavis.edu.

Thank you,
David

andyleynes, 2024-07-23:
David,
I sent data to your email. Something else I learned. I have 8 different
epoched EEG files at the end. If I select multiple EEG files to clear, the
equations in the channel operations window clear and present a red error
message that the channels must match. This disappears when you clear the
files.

Also the loss of bin information happens if you first delete 1 of the 8 EEG
files and then delete the rest with one click. The key seems to be that
first click on clear EEG file with multiple files. I didn't get the error
with only 2 EEG files open.

guanghuizhang0328, 2024-07-25:
> David, I sent data to your email. Something else I learned. I have 8 different epoched EEG files at the end. If I select multiple EEG files to clear, the equations in the channel operations window [...]

Hi,

This (i.e., a red error message) will occur  if the number of channels or bins varies across the selected EEGsets/ERPsets.  We check these information because we allow to process multiple EEGsets/ERPsets. Hope this will help.

Best,
Guanghui