ucdavis/erplab #196: create event list not active

jeenalrambhia, 2024-08-07:
I loaded the set file but i am not able to create event list as the option is greyed out also my events are non numeric class but am not able to convert it to numeric value as the option is greyd out. please help me understand the mistake if i am doing something wrong,

[image]

davidg622, 2024-08-07:
Hello,

The fastest way for me to help you would be for you to email me the data you are using so I can try and figure out what might be causing the issue. You can send the data to my email at drgarrett@ucdavis.edu

Thanks,

David

jeenalrambhia, 2024-08-08:
Thanks for the quick response the data is a bit large so sharing in google
drive link. Any help on this is highly appreciated.

https://drive.google.com/drive/folders/1kzLGgYIi9NcjO-UG88UeYoyjtRKv18gM?usp=sharing

Thanks
Jeenal Rambhia

davidg622, 2024-08-08:
Hi Jeenal,

It seems that you used EEGLAB's epoching tools before attempting to create an EVENTLIST. Please note that the EVENTLIST functions require unepoched data to work correctly.

Best regards,
David

stevenjluck, 2024-08-08:
If you don’t still have the continuous data, you can convert the epoched data back to continuous data. See our FAQ page for details: https://github.com/lucklab/erplab/wiki/Troubleshooting-and-Frequently-Asked-Questions#i-have-already-epoched-my-data-in-eeglab--is-it-possible-to-add-an-eventlist-to-a-dataset-that-was-epoched-using-eeglab