ucdavis/erplab #198: Unable to create a BESTset with ERPLAB Classic

Lee666666, 2024-11-20:
### Description
I am trying to practice decoding EEG data using ERPLAB Classic. The data I am using is provided by https://doi.org/10.18115/D5KS6S. I have encountered an issue that I cannot create a BESTset file, and Matlab showed some information on the errors in the command window. What should I do to fix this issue? Thanks for any useful help.  
[image]
[image]

[*Description of the bug or feature*]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |
| ERPLAB version | [*fill in*] |

Kwinsler, 2024-11-20:
Hi Lee, what version of MATLAB, EEGLAB, and ERPLAB are you using? If you are using a version of MATLAB earlier than 2020b, it is likely that this is the issue.

Lee666666, 2024-11-21:
> Hi Lee, what version of MATLAB, EEGLAB, and ERPLAB are you using? If you are using a version of MATLAB earlier than 2020b, it is likely that this is the issue.

Thanks so much. I'm using MATLAB 2019b. According to your advice, I'll change a later version of MATLAB.

closed 2024-11-21 by Lee666666