ucdavis/erplab #199: Incorrect use of eegfilt, epochframes must be at least 3 times the filtorder

PengfeiLu1, 2024-12-05:
### Description

[*epochframes must be at least 3 times the filtorder. When use the estudio toolbox to create a BEST file and transform the data into phase-independent power, an error message will be displayed : epochframes must be at least 3 times the filtorder.*]

1. Load my data which have been preprocessed by esudio

> My preprocessing steps include downsampling to 500 Hz, bandpass filtering from 0.1 to 30 Hz using the 'eegfilternew' function, performing bin-based epochs, whole brain average re-referencing, ICA removes blink artifacts, and removal of bad epochs exceeding ± 100 μV.

2. Extract Bin-Epoched Single Trials (BEST)

> Convert the data to BEST format and select 'transform into phase-independent power'

[image]

3. Run and Error

> I clicked RUN, and then the following error appeared.

“eegfilt(): filter order is 1500. Misuse eegfilt
epochframes must be at least 3 times the filtorder.

ERROR pop_extractbest (line 291 )
        filtData(:,c,:) =abs(hilbert(eegfilt(squeeze(unfiltData(:,c,:)),fs_original,bandpass_freq(1,1),bandpass_freq(1,2))')').^2;   %Instantaneous power

ERROR f_EEG_extr_best_GUI/avg_run (line 654 )
            [BEST,LASTCOM] = pop_extractbest( EEG , 'Bins', binselec, 'Criterion', artcritestr,'DSindex',1, ...
 
Error calculating UIControl callback.”

I am expecting the smooth completion of this BEST operation, but it prompts the above error. I don't know how to adjust my parameters to ensure normal operation

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windos 11*] |
| Matlab version | [*2022b*] |
| EEGLAB version | [*2023.1*] |
| ERPLAB version | [*12.00*] |
| ERPLAB Studio version | [*12.00*] |

stevenjluck, 2024-12-05:
Hi. Thanks for reporting this.

I’m pretty sure the problem is that your epoch length is not long enough to estimate the power between 1 and 4 Hz. To estimate the power at 1 Hz, the epoch must be long enough to have multiple cycles of a 1 Hz oscillation. Typically, one would want an epoch that has at least 5 cycles of the lowest frequency in your frequency range (5 seconds in your case given that your lowest frequency is 1 Hz). This is not really a limitation of the software – it is a limitation of the underlying mathematical principles. But the software is giving you an error message because the data violate the assumptions of the math.

So, I recommend re-epoching your data with a 5 second epoch length and then trying to extract the bin-epoched single trials from these epochs.

Steve

image.png (view on web)<https://github.com/user-attachments/assets/a5063ac5-e68e-44b5-84d3-c8f1d130d2ad>

PengfeiLu1, 2024-12-05:
Thank you very much for answering my question!

I also think it's because my data length is not enough, my epoch is [-600, 900ms], but it's already the longest I can do. Due to the limitations of the experimental paradigm, I am unable to ensure that each epoch has a length of 5 seconds without overlapping.

So, I still have some questions. Does this mean that my data is not suitable for conversion using this method to transform trials into phase-independent frequency-band EEG power？

I aim to achieve decoding analysis of power in the 1-4Hz frequency band.

As far as I know, Estudio defaults to using FIR filters. If I replace other filters, like IIR, can I achieve a 1-4 Hz filtering conversion? Or rather, based on my current data, I cannot achieve 1Hz filtering even if I change the filter.

Thank you again for your sincere help！

PengfeiLu

stevenjluck, 2024-12-05:
Hi PengfeiLu. This is an issue of the fundamental nature of oscillations, not an issue of what filters are being used. In an ideal world, you need at least 2 seconds of data for the amplitude of a 1 Hz oscillation to be identified. Some filters do a better job of working close to the theoretical minimum epoch length than others, and the filter we use when we create BESTsets is very good in this regard. With a 1500 ms epoch length, you just can’t estimate power at 1 Hz. The best you could do would probably be 3 Hz. But keep in mind that filtering reduces temporal resolution, so any activity you saw at, for example, 100 ms poststimulus, would be impacted by signals that actually occurred at -500 ms and +700 ms.

Steve