ucdavis/erplab #203: pop_artmwppth/eegplot fails to update marker removal on review

mjowens, 2025-02-18:
After running  pop_artmwppth* any epoch with an artifact cannot be removed after it calls pop_eegplot for review. Typically, in the eegplot clicking "update markers" will result in a pop message that warns you about marking epochs when removing or adding epochs. Now it only shows the warning message when adding epochs and will not register (or allow) any change to remove highlighted epochs.

In addition to using the gui have tried to the following.
1. using the legacy eegplot
2. running EEG = eegcheckset( EEG) after pop_artmwppth
3. running eeglab redraw pop_artmwppth, and either: saving over previous set and creating a new one
4. adjusting the flag codes

*I have observed this issue using in-line code (below) and the GUI interface.
      EEG  = pop_artmwppth( EEG , 'Channel', [1:29], 'Flag', [1 8], 'Review', 'on', 'Threshold',  100, 'Windowsize',  200, 'Windowstep',  100 );

Versions
MATLAB R2024a
ERPLAB 11.03

davidg622, 2025-02-19:
Hello,

Just to provide clarification on the issue you bring up:
After running pop_artmwppth, a plot will open, allowing you to mark/unmark epochs for rejection. After pressing "Update Markers", the following popup will appear:

[image]

Is this the same warning message that you are referring to?

David

mjowens, 2025-02-20:
> Hello,
>
> Just to provide clarification on the issue you bring up: After running pop_artmwppth, a plot will open, allowing you to mark/unmark epochs for rejection. After pressing "Update Markers", the [...]

The pop-up will appear only when adding marks to epoched data. When trying to remove marked epochs no pop-up will appear and changes will not be updated.

davidg622, 2025-02-23:
Hello,

I attempted to replicate your issue with my own data, but found that in unmarking epochs, my data was updated as expected. There are two things we might try now:
1. What version of EEGLAB are you using? The most recent version is v2024.2. If it's not up to date, that might be causing the issue.
2. You could email me a sample of your data at my drgarrett@ucdavis.edu, and I can try to replicate this issue on your data.

Best,

David

mjowens, 2025-02-25:
Thank you much David, 

we have been using EEGLAB v2024.1. We updated to that version near the time it came out and before processing this round of data. So everything is quite recent. Nothing is jumping out to me code-wise. I would like to know why its not working now after several years of code working with older versions before updating again. But, I will try an update fix if needed after your review. I'll send you a file later today. 

Thanks again.

davidg622, 2025-02-26:
Hello,

I attempted to replicate your issue with your data, and the marking/unmarking functionality seems to work just fine. Given that it does not seem to be MATLAB, EEGLAB, or ERPLAB or your data that is the problem, my next thought might be to ask you what OS you are using. If that does not end up helping, we might want to schedule a Zoom meeting so I can see what happens live.

David

mjowens, 2025-02-26:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

Thanks David, that's odd. We see it happening on separate PC's at different universities. We are using Windows OS.

[EXTERNAL EMAIL] DO NOT CLICK links or attachments unless you recognize the sender and know the content is safe.

mjowens, 2025-03-12:
Hi David, would it still be possible to set-up a zoom or teams meeting sometime? After updating EEGLAB and ERPLAB I still notice the error, with a caveat.

I have been using file: MIMIC3_PV (1 face) w EEG-6153-1_EEBP_ARTDET. In trying different combinations of marking and unmarking, I noticed that the unmarking will only update for one epoch, if I unmark more than one epoch nothing will happen — that is no pop-out warning for updating the marks will appear. If this week doesn't work for you, I am free across next week to meet during our spring break.

Thanks for you help so far,

Max