ucdavis/erplab #204: ERPLAB->AVERAGE ACROSS ERP SETS(GRAND AVERAGE) error

Rosie9417, 2025-03-06:
When I tried to plot the grand average using erplab as title describes, it bumped into the error:

Unrecognized function or variable 'rms'.

Error in gaverager (line 458)
            dq.data = rms(dq_data_here,4);

Error in pop_gaverager (line 352)
[ERP, serror, msgboxText ] = gaverager(ALLERP, lista, optioni, erpset, nfile, wavg, stderror, artcrite, exclunullbin, warn, DQ_spec);

Error in pop_gaverager (line 173)
            [ERP, erpcom]  = pop_gaverager(ALLERP, 'Erpsets', erpset,'Criterion', artcrite, 'SEM', stdsstr,...
 
I am not sure what is the problem and how to proceed. It would be great if there is any way that I can proceed.
Thank you.

Very best,
Yesol

Kwinsler, 2025-03-06:
Hi Yesol,

What versions of MATLAB, EEGLAB, and ERPLAB are you using?

At first glance, it looks like your MATLAB environment is missing the rms (root mean square) function. This could be tested directly by just calling this function in your MATLAB command window, e.g.:
`test = rms([1 2 3 4]);`

If you get the same error (Unrecognized function or variable 'rms'.), then you could likely solve this issue by updating to the latest version of MATLAB. Before MATLAB 2022a, the rms function was in the signal processing toolbox (this toolbox is still required for ERPLAB to function properly). 

Please let us know if this does not fix the issue.

If for some reason this is not possible or does not fix the issue, and you need a quick work around:

The rms function is only called if you wish to get information about data quality. If you don't want this information, you can disable this by setting the 'DQ_Flag' argument to 0 when you run pop_gaverager. e.g.:

`pop_gaverager(ALLERP, 'Erpsets', erpset,  'DQ_Flag', 0, (other arguements...) )`

I hope this helps, let us know how it goes!

-Kurt

Rosie9417, 2025-03-07:
Hi Kurt,

Thank you for your assistance! Previously, the Matlab that I have been using had not access to signal processing tool although it was downloaded. I resolved it by updating Matlab and re-downloading signal processing toolbox. 

Very best,
Yesol