ucdavis/erplab #205: INQUIRY: How to adjust the scale bar and see the units in the ERP waveforms?

ixl63, 2025-03-30:
Hello, I am a beginner with ERPLab and I am struggling to find how to see the units of the X-axis and Y-axis and adjust the scale. When I try to enlarge an individual ERP waveform it doesn't let me do so. I want to confirm that the Y-axis is in uV. Please advise! Thank you!

Kwinsler, 2025-04-01:
Hello,

Yes, the Y-axis is in uV (microvolts) and the X-axis is in ms (milliseconds). 

If you are using the ERPLAB gui, after you select Plot ERPs>Plot ERP waveforms:

The X and Y scales can be changed in the left panel labeled "scales".

The X axis is the time range (in ms), and you can specify this range. If you wish to specify the tick marks, you have to uncheck the "auto time-ticks" check box and then can click the "time ticks" button to specify them. 

The Y axis is voltage (in uV). You can specify this range by unchecking the "auto Y-scale" check box, and then you can enter the range you would like (the min and max voltage). If you wish to specify the tick marks, you have to uncheck the "auto Y-ticks" check box and then can click the "Y ticks" button to specify them. 

If you are using a script, here is an example of how to plot an ERP while specifying the scale and tickmarks for both the Y and X axes:

```
%% Specify some parameters
% note these values will need to be changed to match your data
bin_list = [37 38]; % List of bins to be plotted
p_chans = [11 10 23 24 12 14 27 26]; % some channels to plot
baseline_period = '-100 0'; %start time and end time of baseline period
xscale = [-100.0 400.0   -100:100:400]; %Set x-axis scale (start time   end time) and tick marks in milliseconds for every 100 ms (start:by:stop)
yscale = [-5.0 5.0   -5:2:5]; %Set y-axis scale (start time   end time) and tick marks in microvolts for every 2 between -5 and 5

%% Plot some ERPs
ERP = pop_ploterps( ERP, bin_list, p_chans, 'blc', baseline_period, 'Maximize', 'off', 'Style', 'Classic', 'xscale', xscale, 'yscale', yscale);
```

I hope this helps, let us know if you still have questions!

-Kurt

ixl63, 2025-04-10:
Hello Kurt, 

Thank you so much! This was very helpful.

closed 2025-04-10 by ixl63