ucdavis/erplab #207: Error when trying to create grand average

ixl63, 2025-04-26:
Hello! I am trying to create a grand average with all of my processed ERPs but I keep getting this error. Would someone mind explaining what it means and how I can resolve this issue? Thank you so much!

[image]

stevenjluck, 2025-04-26:
Hi. This error message is indicating that the number of sample points in each waveform in ERPsets 25 and 26 is different from the number of sample points in the other ERPsets. To average ERPsets together, the number of time points prior to time zero and the number of time points after time zero must be the same in all waveforms.

This problem may indicate that you specified a different epoch length for ERPsets 25 and 26 compared to the other ERPsets, or that the sampling rate was different.

Steve

image.png (view on web)<https://github.com/user-attachments/assets/8a2c8fc7-fa6d-4ea3-a06c-87eed57e252e>

ixl63, 2025-04-26:
Hi Dr. Luck, thank you for your response!

I went through and double-checked the different epoch lengths (-200 to 800) as well as the sampling rates (all at 250Hz). I reprocessed a few of the files that were coming up in the error but the message kept going down the list.

[image]
It would go from an error with 125 and 126 --> 124 and 125 --> 123 and 124, and it would continue on in this pattern. Do you have an idea as to what I can do to fix this error?

stevenjluck, 2025-04-26:
I recommend taking a close look at the sampling rate and epoch length for all of your ERPsets. For example, the error message you provided indicates that ERPset 23 has a sampling rate of 100 Hz.

image.png (view on web)<https://github.com/user-attachments/assets/74a0d548-3052-4592-9bf1-3018ccf857f7>

ixl63, 2025-04-26:
Hi Dr. Luck, I just learned that I can select all the files at once and reapply the sampling rate in the ERP tab of EStudio. It works now, thank you so much!

closed 2025-04-26 by ixl63