ucdavis/erplab #210: make_SEM_set does not match hand-calculated SEM

lianapabmeir, 2025-07-17:
We are attempting to use make_SEM_set to calculate trial-to-trial variability in amplitude within each participant. However, we are unsure how ERPLAB is calculating this SEM. First code block (below) includes the call to make_SEM_set.

However, when hand-calculating SEM, values consistently do not match what we get in SEM_erp.binerror or SEM_erp.bindata. Second code block shows hand-calculation method/check. We cannot identify any issues with this way so far (subsetting/indexing appears correct). 

Any clarification on make_SEM_set would be appreciated.

Code block 1 (use of make_SEM_set)
```
for j = 1:length(P)
    eeg = pop_loadset('filename',[P{j} '_preprocessed.set'],'filepath','/Users/Downloads/SH_files_no_detrend/Preprocessed/');
    eeg = eeg_checkset( eeg );
    
    % fixes trigger latency
    for i = 1:length(eeg.event)
        eeg.event(i).latency = eeg.event(i).latency+(200/2);
    end

    eeg  = pop_creabasiceventlist( eeg , 'AlphanumericCleaning', 'on', 'BoundaryNumeric', { -99 }, 'BoundaryString', { 'boundary' }, 'Eventlist',...
        ['/Users/Downloads/SH_files_no_detrend/ERPs/' P{j} '_conds.txt'] );

    eeg  = pop_binlister( eeg , 'BDF', '/Users/Downloads/SH_files_no_detrend/MMN_BioSemiCodes.txt', 'ExportEL',...
        ['/Users/Downloads/SH_files_no_detrend/ERPs/' P{j} '_elist.txt'], 'ImportEL', ['/Users/Downloads/SH_files_no_detrend/ERPs/' P{j} '_conds.txt'],...
        'IndexEL',  1, 'SendEL2', 'EEG&Text', 'UpdateEEG', 'on', 'Voutput', 'EEG' );

    eeg  = pop_epochbin( eeg , [-50.0  330.0],  'pre');
    eeg  = pop_artextval( eeg , 'Channel',  1:32, 'Flag',  1, 'Threshold', [ -100 100], 'Twindow', [ -50 330] );        
    eeg  = pop_basicfilter( eeg,  1:32 , 'Boundary', 'boundary', 'Cutoff',  20, 'Design', 'butter', 'Filter', 'lowpass', 'Order', 2 );
    eeg = pop_saveset(eeg, [erpFolderPath P{j} '_before_ave_erp']); % added 6/16

    erp = pop_averager( eeg , 'Criterion', 'good', 'ExcludeBoundary', 'on', 'SEM', 'on' );

    erp = pop_binoperator(erp, { 'bin4=b2-b1'}); % add this to see MMN

    erp = pop_savemyerp(erp, 'erpname', [P{j}], 'filename', [P{j} '.erp'], 'filepath', '/Users/Downloads/SH_files_no_detrend/ERPs/', 'Warning', 'off');

    SEM_erp = make_SEM_set(erp,'gui',1);        
    SEM_erp = pop_savemyerp(SEM_erp, 'erpname',...
        [P{j} '_SEM'], 'filename', [P{j} '_SEM.erp'], 'filepath', '/Users/Downloads/SH_files_no_detrend/ERPs/', 'Warning', 'off');   
        
end
```

Code block 2 (hand-calculation/comparison of SEM)
```
sem_erp = pop_loaderp('filename','1M_SEM.erp','filepath','/Users/Downloads/SH_files_no_detrend/ERPs/');
data = sem_erp.EVENTLIST.eventinfo; % get the event data
labels = {data.codelabel};
condition2_idx = strcmp(labels, 'condition2');
condition2_items = [data(condition2_idx).item]'; % just extract the trial numbers for the deviant trials (bin 2)

eeg = pop_loadset('filename','1M_before_ave_erp.set','filepath','/Users/Downloads/SH_files_no_detrend/ERPs/');
subset = eeg.data(:, :, condition2_items);
sem_hand_calculated = std(subset, 0, 3) ./ sqrt(size(subset, 3)); % third dimension is number of trials

bin2_data = sem_erp.bindata(:, :, 2); % just looking at bin 2 data

subtraction = sem_hand_calculated - bin2_data; % comparison, does not = 0
```

Kwinsler, 2025-07-17:
Hello,

Thanks for reporting this potential issue. I won't be able to replicate this exactly without copies the files you are using (and the version of ERPLAB you are using). But looking at your code, I can notice one possible difference between the ERPLAB calculated SEM values and your hand calculated values. That is, the SEM calculation by pop_averager excludes trials with artifacts, while it does not appear that your own calculation excludes trials with artifacts. So, if there are any trials with artifacts, this would cause some difference in the SEM between the two calculations.

To be sure that there is a difference between the two calculations, we would need to first verify that the exact same data is being used (the same exact trials, time windows, electrodes, etc.). If you can verify this and there is still a difference, please let us know and we can work to figure out whats going on.

Thanks,
-Kurt

lianapabmeir, 2025-07-22:
Kurt, thank you for the quick response. You are exactly right -- it's clear that this hand-calculation included trials with artifacts. Excluding trials with artifacts resolved the discrepancy.